circuitikz/circuitikz #197: Power electronics library

mateuspinheirodias, 2019-05-08:
Hi, guys.

I´m a power electronics master student at UNICAMP-Brazil,  and just discovered circuitikz. 
I'm think about create a PE library with the basic converters: DC-DC(buck , boost and buck-boost), AC-DC and DC-AC, and solar and wind energy. 
If i succeed , i want to share the library with you. 

Greeting from brazil.

Mateus Pinheiro Dias

Rmano, 2019-05-09:
Great! Please have a look at https://github.com/circuitikz/circuitikz/blob/master/CONTRIBUTING.md before, and be sure to check out the latest version to work on.

Rmano, 2020-05-02:
There have been nice addition in this field by @olfline.
Check them out: #396!

Rmano, 2021-01-06:
Closing this issue due to inactivity; please feel free to reopen if needed.

closed 2021-01-06 by Rmano