circuitikz/circuitikz #201: Op amp doesn't honor color setting

titizaki, 2019-05-15:
Hello,
Some of my documents are white text on dark background. Circuitikz works very well with this, except for one detail: the + and - signs for the inputs of a operational amplifier. They ignore all settings and remain black.

The following minimal example shows the problem: in the output, the op amp is drawn in white as expected, except the signs.

Thanks for your attention.

```
\documentclass{article}
\usepackage{pagecolor}
\pagecolor{darkgray}
\color{white}
\usepackage{circuitikz}
\tikzset{color=white}

\begin{document}

\begin{circuitikz}
\draw (0,0) node[op amp](ao){};;
\end{circuitikz}

\end{document}
```

titizaki, 2019-05-15:
Thanks for your quick answer. I made sure xcolor was loaded first, it didn't change anything. As for the \color{white} command, it has to be in the preamble in my use case, since it's in fact in the class file. But I checked, it isn't cause.

But I did find a big clue: I use XeLaTeX. If I fall back to PDFLaTeX, then the color is correctly applied to everything.

I don't know how to study this problem, but perhaps it's related to this one:

[https://tex.stackexchange.com/questions/253029/font-color-is-always-black-in-mdframed](url)

It's mentioned in the last answer in the page, the post by Keith Davies.

Since the problem is probably not in Circuitikz then, do you have any advice to work around the problem? Can I manually change the color of both signs manually? Or fill the triangle shape with a lighter color so that the signs remain visible?

Thanks for your attention.

sistlind, 2019-05-16:
Do you really have to use xelatex? Maybe using Lualatex is an option for
you?

Rmano, 2019-05-16:
I can't check now (I'm commuting), but you can try to set the color explicitly in
```
\ctikzset{tripoles/op amp/font=\fontsize{10} {12} \selectfont\boldmath\color{white}}
```

I will try to reproduce it and see if I can find a workaround.

titizaki, 2019-05-16:
About switching to LuaLaTeX: at this point it's not possible. XeLaTeX is vastly superior when it comes to font support, and I need it for bilingual documents. But I know both compilers improve steadily, so I'll probably compare them again at some point.

Your workaround with \ctikzset works perfectly. I added to my class file, and now everything is back to normal. Thank you. :)

Rmano, 2019-05-16:
Good to know. 

But thinking on it... the plus and minus are part of the symbol, so it would be better to draw them with lines instead of using math formulas... that way they will scale with the symbol. @sistlind @mredaelli @der-stefan , what do you think?

sistlind, 2019-05-20:
As +/- are math signs I think its better to draw them as text.

Rmano, 2019-05-21:
> As +/- are math signs I think it's better to draw them as text.

Yes, I was trying some experiment and scaled-up "+" and "-" are quite... ugly. I'll close this issue now. Thanks!

closed 2019-05-21 by Rmano