circuitikz/circuitikz #211: `yscale=-1` causes incorrect drawing of bipoles

asinghvi17, 2019-05-29:
MWE:
```latex
\RequirePackage{luatex85}
\documentclass[tikz, export=png]{standalone}

\usepackage{circuitikzgit}

\begin{document}

\begin{tikzpicture}
  \draw (0, 0) to [R] (0, 2);
  \draw (1, 2) to [R] (1, 0);
\end{tikzpicture}

\begin{tikzpicture}[yscale = -1]
  \draw (0, 0) to [R] (0, 2);
  \draw (1, 2) to [R] (1, 0);
\end{tikzpicture}

\end{document}
```
The first `tikzpicture` has no `yscale=-1`, and the resistors are both drawn normally.  However, in the second, there is a vertical line through both resistors; I suspect this is because the resistor are being drawn without the `yscale=-1` option, even when it is set globally.

Is there any change to `circuitikz` that could fix this, or should I just work with no y-flip, on the negative y-axis?

Rmano, 2019-05-29:
Hmmm... the problem is that `circuititk` is putting the "left" anchor to the first node, and the "right" anchor to the second one; when `yscale=-1`  is active, the coordinates are swapped, but the nodes are not rotated, so the point above is connected to the point below and the other way around. You can check this because adding `tranform shape` to the options all is ok, but now the labels are mirrored, too.

This behavior was the same in at least 0.8.3, so it's not a regression, at least. 

It's a bug, but I am not sure on how to tackle it.

asinghvi17, 2019-05-29:
Thanks, this isn't high priority at all so don't worry if it takes some time.  I actually converted my breadboard layout to use negative y which is actually more intuitive for displacements.

Rmano, 2019-05-30:
This is really complex, and it goes all way back at least to 0.6. I am trying to understand exactly what's happening, but touching the main code in `pgfcircpath` is more difficult than expected, and it has a bad effect on labels, too. If you like to dive in, I have a test branch on my clone: https://github.com/Rmano/circuitikz/tree/bug-211  

To see real fireworks, try to play with: 

```
\documentclass[]{article}

\usepackage{circuitikz}

\begin{document}

\begin{tikzpicture}
  \draw (0, 0) to [D, l=D1] (0, 2);
  \draw (0, 0) to [D, l=D2] (2, 0);
  \path (0,0) node[draw]{A};
  \path (0,2) node[draw]{B};
  \path (2,0) node[draw]{C};
\end{tikzpicture}

\bigskip

\begin{tikzpicture}[
        % rotate=45,
        yscale = -1,
        % every node/.style={transform shape}
    ]
  \draw (0, 0) to [D, l=D1] (0, 2);
  \draw (0, 0) to [D, l=D2] (2, 0);
  \path (0,0) node[draw]{A};
  \path (0,2) node[draw]{B};
  \path (2,0) node[draw]{C};
\end{tikzpicture}

\end{document}
```

So for now I conclude that `circuitikz` is not working with global negative `scale` and with `rotation` parameters... 

I will leave this here, opened, in the hope that I can find the time to rewrite all the code

Rmano, 2019-06-02:
Also, `xscale=1` and `yscale=2` is funny (was there in 0.4, checked).

The problem is the `\pgf@circ@direction` calculation, and all the anchor shuffling; changes will affect almost everything...

I will merge a documentation update for now, see PR #217

asinghvi17, 2019-06-13:
Also, @Rmano, you can probably make branches on the main repo as a collaborator, right?  Helps with discovery, and you don't need to resync your fork's master every time you want to make a PR :)

Rmano, 2019-06-13:
I prefer (probably because my git skills are quite basic) to branch on my fork to test things... and let the main repository clean. I fear that it could become like some other projects I'm in and that have grown to a forest, really...

Rmano, 2019-06-16:
A lot of time ago there was a `develop` branch --- to split maintenance and new features. Maybe I can try to do it again, with topic branches... will study a bit more `git` and I'll see.

Rmano, 2019-07-29:
Asked a question on tex.stackexchange.com https://tex.stackexchange.com/questions/501929/tikz-angle-calculation-and-rotation-with-transformed-coordinates

Rmano, 2019-07-30:
Well, with the help of @marmot of tex.stackexchange.com, I prepared a patch that _seems_ to fix this issue. It's a quite scary thing, changing a basic part of the package, but... the problem is that it now breaks everything if `transform shape` is active. Will try to experiment a bit more.

Rmano, 2019-07-30:
Ok, I hacked something out. With this change, the original snippet renders as this: 

```
\documentclass[]{article}
\usepackage{circuitikz}
\begin{document}
\begin{tikzpicture}
  \draw (0, 0) to [R=R] (0, 2);
  \draw (1, 2) to [R=R] (1, 0);
\end{tikzpicture}

\begin{tikzpicture}[yscale = -1]
  \draw (0, 0) to [R=R] (0, 2);
  \draw (1, 2) to [R=R] (1, 0);
\end{tikzpicture}
\end{document}
```
renders like: 

[image]

which seems correct. Moreover, the longish example:

```latex
\documentclass[]{article}

\usepackage{circuitikz}

\begin{document}

\begin{tikzpicture}
  \draw (0, 0) to [D, l=D1, a=A1] (0, 3);
  \draw (0, 0) to [D, l=D2, a=A2] (3, 0);
  \draw (0, 0) to [D, l=D3, a=A3] (3, 3);
  \path (0,0) node[draw]{A};
  \path (0,3) node[draw]{B};
  \path (3,0) node[draw]{C};
\end{tikzpicture}

\bigskip
\typeout{Now the scaling one:}
\begin{tikzpicture}[
        % rotate=30,
        xscale = 2,
        % yscale = -1,
        % scale = 2,
        % transform shape,
        % every node/.style={transform shape}
    ]
  \draw (0, 0) to [D, l=D1, a=A1, ] (0, 3);
  \draw (0, 0) to [D, l=D2, a=A2, ] (3, 0);
  \draw (0, 0) to [D, l=D3, a=A3, ] (3, 3);
  \path (0,0) node[draw]{A};
  \path (0,3) node[draw]{B};
  \path (3,0) node[draw]{C};
\end{tikzpicture}
\typeout{Now the rotating one:}
\begin{tikzpicture}[
        rotate=30,
        % xscale = 2,
        % scale = 2,
        % transform shape, every node/.style={transform shape}
    ]
  \draw (0, 0) to [D, l=D1, a=A1, ] (0, 3);
  \draw (0, 0) to [D, l=D2, a=A2, ] (3, 0);
  \draw (0, 0) to [D, l=D3, a=A3, ] (3, 3);
  \path (0,0) node[draw]{A};
  \path (0,3) node[draw]{B};
  \path (3,0) node[draw]{C};
\end{tikzpicture}

\begin{circuitikz}[american, rotate=-20, 
        transform shape,
        ]
    \draw (0,0) -- ++(1,0) to[R=R] ++(2,0)
    to [rmeterwa, t=A, i=$i$] ++(0,-2) node[ground]{};
    \draw (1,0) to[rmeterwa, t=V, v=$v$] ++(0,-2) node[ground]{};
\end{circuitikz}

\end{document}
```

produces 

[image]