circuitikz/circuitikz #213: Discussion issue: Positioning of DIP chips on a breadboard layout

asinghvi17, 2019-05-29:
I want to use `circuitikz` to make a breadboard layout.  
Preamble:
```latex

\tikzset{
  open socket/.style = {
    circle,
    fill = lightgray,
    inner sep = 1pt
  },
  filled socket/.style = {
    circle,
    fill = black,
    inner sep = 1pt
  }
}
\usetikzlibrary{calc, arrows.meta, backgrounds}
```
So far, I've gotten the grid:
```latex
%% draw breadboard sections
  \foreach \y [evaluate = \y as \line using int(abs(\y))] in {0, ..., -13}{

    % draw line numbers off to the left
    \node [text=gray] at (-9, \y) {\line};
    \node [text=gray] at (9, \y) {\line};

    % draw bus and breadboard grid sockets
    \foreach \lab/\x in {l-/-8, lg/-7, A/-5, B/-4, C/-3, D/-2, E/-1, F/1, G/2, H/3, I/4, J/5, rg/7, r+/8} {
      \coordinate [open socket] (\lab\line) at (\x, \y);
    }

    % draw row connectivity for board proper
    \draw [lightgray, opacity = 0.2, very thick]
      (-5, \y) -- (-1, \y)
      (1, \y)  -- (5, \y);

  }

  % label breadboard columns

  \foreach \lab/\x in {A/-5, B/-4, C/-3, D/-2, E/-1, F/1, G/2, H/3, I/4, J/5} {
    \node [above of = \lab0] {\lab};
  }

  % label buses

  \node [above of=l-0] {V$_{-}$};
  \node [above of=lg0] {GND$_{}$};
  \node [above of=rg0] {GND$_{}$};
  \node [above of=r+0] {V$_{+}$};

  % draw bus connection lines
  \begin{scope} [
      on background layer  % these paths should not interfere with anything else
    ]
    \draw [very thick, red,  opacity = 0.2] (l-0) -- (l-13);
    \draw [very thick, blue, opacity = 0.2] (lg0) -- (lg13);

    \draw [very thick, blue, opacity = 0.2] (rg0) -- (rg13);
    \draw [very thick, red,  opacity = 0.2] (r+0) -- (r+13);
  \end{scope}
```
[image]
Now, I want to position a DIP chip in the middle of the breadboard, and what I've got till now is this:

```latex
\node at (E1) (081) [
                        dipchip,
                        fill = cyan!20!white,
                        circuitikz/multipoles/dipchip/pin spacing = 0.711,
                        circuitikz/multipoles/dipchip/width = 1.03,
                        anchor = pin 1,
                        num pins = 8,
                        hide numbers
                    ] {TL081};

  % and fill its pins with tiny descriptions:

  \node [right, font=\tiny] at (081.bpin 1) {ON1};
  \node [right, font=\tiny] at (081.bpin 2) {IN-};
  \node [right, font=\tiny] at (081.bpin 3) {IN+};
  \node [right, font=\tiny] at (081.bpin 4) {$V_-$};

  \node [left,  font=\tiny] at (081.bpin 5) {ON2};
  \node [left,  font=\tiny] at (081.bpin 6) {OUT};
  \node [left,  font=\tiny] at (081.bpin 7) {$V_+$};
  \node [left,  font=\tiny] at (081.bpin 8) {NC};

  \foreach \i in {1, 2, ..., 8} {
    \node at (081.pin \i) [filled socket] {};
  }
```
which yields:
[image]
The _result_ itself is fine; however, it requires that I manually align the chip, which isn't very flexible.  Is there a better way to do this alignment, or could one be made?  Alternatively, is there a spacing method that I can use for the breadboard that will allow the chip spacing to correspond to it?

Thanks, and sorry for the long winded issue.

Rmano, 2019-05-29:
You can play with the fact the in `circuititz` all the lenghts are proportional to `bipoles/lenght` (default 1.4cm); for example:

```
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\ctikzset{bipoles/length=1cm}
\ctikzset{multipoles/dipchip/pin spacing=1}
\ctikzset{multipoles/external pins width=0.5}
\ctikzset{multipoles/dipchip/width=3}
\begin{document}
\begin{circuitikz}[x=1cm, y=1cm,
    ]
    \draw [cyan, dashed](-4,-4) grid (4,4);
    \draw (-3,3) to[R, *-*] ++(1,0);
    \path (-2,3) node[dipchip, anchor=pin 1, num pins=12]{DIP};
\end{circuitikz}
\end{document}
```

gives: 

[image]

asinghvi17, 2019-05-30:
Would it make sense for me to try and add this to the manual?  Perhaps in the `Examples` section, to show the workarounds required for breadboard design?

asinghvi17, 2019-05-30:
Also, I just fixed my example with the second trick you posted on StackExchange [here](https://tex.stackexchange.com/questions/493239/alignment-of-dip-chips-in-circuitikz/493255#493255), the example now looks like this:

```latex
\documentclass[tikz]{standalone}

\usepackage[siunitx]{circuitikzgit}

\usepackage{xcolor}

% \ctikzset{bipoles/length=1cm}

\usetikzlibrary{calc, arrows.meta, backgrounds}
\tikzset{
  open socket/.style = {
    circle,
    fill = lightgray,
    inner sep = 1pt
  },
  filled socket/.style = {
    circle,
    fill = black,
    inner sep = 1pt
  }
}

% work with dipchips as well
\makeatletter\newlength{\Rlen}\setlength{\Rlen}{\pgf@circ@Rlen}\makeatother
\ctikzset{multipoles/dipchip/pin spacing=1}
\ctikzset{multipoles/external pins width=0.5}
\ctikzset{multipoles/dipchip/width=1}

% \ctikzset{multipoles/dipchip/pin spacing=1}

\begin{document}

\begin{tikzpicture}[
    % scale = 1,
    % yscale = -1,  % on breadboards, the y axis points downward
    x=\Rlen, y=\Rlen,
  ]

  %% draw breadboard sections
  \foreach \y [evaluate = \y as \line using int(abs(\y))] in {0, ..., -13}{

    % draw line numbers off to the left
    \node [text=gray] at (-9, \y) {\line};
    \node [text=gray] at (9, \y) {\line};

    % draw bus and breadboard grid sockets
    \foreach \lab/\x in {l-/-8, lg/-7, A/-5, B/-4, C/-3, D/-2, E/-1, F/1, G/2, H/3, I/4, J/5, rg/7, r+/8} {
      \coordinate [open socket] (\lab\line) at (\x, \y);
    }

    % draw row connectivity for board proper
    \draw [lightgray, opacity = 0.2, very thick]
      (-5, \y) -- (-1, \y)
      (1, \y)  -- (5, \y);

  }

  % label breadboard columns

  \foreach \lab/\x in {A/-5, B/-4, C/-3, D/-2, E/-1, F/1, G/2, H/3, I/4, J/5} {
    \node [above of = \lab0] {\lab};
  }

  % label buses

  \node [above of=l-0] {V$_{-}$};
  \node [above of=lg0] {GND$_{}$};
  \node [above of=rg0] {GND$_{}$};
  \node [above of=r+0] {V$_{+}$};

  % draw bus connection lines
  \begin{scope} [
      on background layer  % these paths should not interfere with anything else
    ]
    \draw [very thick, red,  opacity = 0.2] (l-0) -- (l-13);
    \draw [very thick, blue, opacity = 0.2] (lg0) -- (lg13);

    \draw [very thick, blue, opacity = 0.2] (rg0) -- (rg13);
    \draw [very thick, red,  opacity = 0.2] (r+0) -- (r+13);
  \end{scope}

  %% We're done with the breadboard!
  %% Now, we need to draw the actual circuit :P

  % Now, I place the op amp:

  \node at (E1) (081) [
                        dipchip,
                        fill = cyan!20!white,
                        % circuitikz/multipoles/dipchip/pin spacing = 0.711,
                        % circuitikz/multipoles/dipchip/width = 1.03,
                        anchor = pin 1,
                        num pins = 8,
                        hide numbers
                    ] {TL081};

  % and fill its pins with tiny descriptions:

  \node [right, font=\tiny] at (081.bpin 1) {ON1};
  \node [right, font=\tiny] at (081.bpin 2) {IN-};
  \node [right, font=\tiny] at (081.bpin 3) {IN+};
  \node [right, font=\tiny] at (081.bpin 4) {$V_-$};

  \node [left,  font=\tiny] at (081.bpin 5) {ON2};
  \node [left,  font=\tiny] at (081.bpin 6) {OUT};
  \node [left,  font=\tiny] at (081.bpin 7) {$V_+$};
  \node [left,  font=\tiny] at (081.bpin 8) {NC};

  \foreach \i in {1, 2, ..., 8} {
    \node at (081.pin \i) [filled socket] {};
  }
\end{tikzpicture}
\end{document}
```
[image]

closed 2019-05-30 by asinghvi17

Rmano, 2019-05-30:
> Would it make sense for me to try and add this to the manual? Perhaps in the `Examples` section, to show the workarounds required for breadboard design?

A contribution to the manual, either in the example or in the tutorial section, would be welcome, of course. Thanks!