circuitikz/circuitikz #215: Discussion: adding old version to the package

Rmano, 2019-06-02:
In my local installation, I have saved some of the old "git releases" and mangled a bit the introduction so I can, for example, do 

```latex
\usepackage{circuitikz-0.8.3} 
```

to use the old version. Is a kind of poor-men (but foolproof) version of the version control that will be in the newer LaTeX. 

Would you think it would be useful to add the files to the package? I have 0.8.3, 0.7, 0.6 now; I would like to add 0.5 and 0.4 (but I can find the tag for the last;  can anyone tell me which commit is that one?)

asinghvi17, 2019-06-02:
@Rmano, this seems to be the commit for version 0.4: https://github.com/circuitikz/circuitikz/commit/e618f578f33ebcf6e1eb79d5e297fad2f063f84c 

as you can see by the git blame at this point: https://github.com/circuitikz/circuitikz/blame/620279f14b09e71d20fd37e0abf76bbc3cccb4d9/tex/latex/circuitikz/circuitikz.sty

Rmano, 2019-06-02:
You can see a tentative pull request in PR #216 (I did not resurrect 0.5 at the end, it was quite complex with 0.4 alone --- I think this is sufficient).

Rmano, 2019-06-04:
Ok, I merged it.

closed 2019-06-04 by Rmano