circuitikz/circuitikz #218: Discussion: relative size of the components

Rmano, 2019-06-06:
Dear `circuitikz` users and developers, I am tackling a little problem with the overall structure of the components. 

The overall problem started with the fact that the relative size of some component  is, for me, really weird: 

[image]

With respect to resistors, diodes are way too big (for me) and transistors are too small... 

I know that I can change them with `\ctikzset`ing the correct parameters, but here the problem arise: unfortunately, the parameters are quite mixed, of different natures, and *not grouped*.  For example: 

```latex
\ctikzset{tripoles/npn/width/.initial=.6}
\ctikzset{tripoles/npn/base height/.initial=.45}
\ctikzset{tripoles/npn/base height 2/.initial=.15}
```
the first parameter is relative to the standard `\pgf@circ@Rlen`, so I have to change it to resize the component... but the other to are relative to the previous size, so I shouldn't touch it. There is no way to automatically distinguish the two types. 
Worst, after that I have to change the `width` and `height` parameter on the other zillions of transistors to keep them consistent... 

What I am doing now is a kind of stop-gap feature, namely this: 

```latex
\def\ctikzscaledim#1#2{% #1: scale factor, #2: parameter
    \edef\otmp{\ctikzvalof{#2}}
    \pgfmathsetmacro{\tmp}{(\otmp)*(#1))}%
    \pgfkeyslet{\circuitikzbasekey/#2}{\tmp}
    % \typeout{SET\space#2\space FROM\space\otmp\space TO\space\tmp}%
}

\def\ctikzscalediodes#1{%
\ctikzscaledim{#1}{bipoles/diode/height}
\ctikzscaledim{#1}{bipoles/diode/width}
\ctikzscaledim{#1}{bipoles/bidirectionaldiode/height}
\ctikzscaledim{#1}{bipoles/bidirectionaldiode/width}
\ctikzscaledim{#1}{bipoles/varcap/height}
\ctikzscaledim{#1}{bipoles/varcap/width}
\ctikzscaledim{#1}{tripoles/thyristor/height}
\ctikzscaledim{#1}{tripoles/thyristor/height 2}
\ctikzscaledim{#1}{tripoles/thyristor/width}
\ctikzscaledim{#1}{tripoles/triac/height}
\ctikzscaledim{#1}{tripoles/triac/width}
}
``` 

so that I can say `\ctikzscalediodes{0.5}` and have: 
[image]

which I like more. I was thinking to go through the pain of this and doing that for transistors too, and maybe amplifiers. But it will be hard to maintain, and I am not so sure if I should push it upstream. 

The solutions could be to do a radical change: group the elements by categories (maybe the ones in the manual?, but better less)  and then do: 

1) change all width, height so that every group use the same parameters. Local changes will be always possible (with scoping), but it would be much easier to have a kind of "stylesheet" for the circuits.  
2) add a kind of scale parameter and invent a `\ctikzvalof` variant which uses the scale parameters per families (for example, \ctikvalof[diodescale]{bipoles/diode/width}` will expand to the width multiplied by the scale. 

1 is cleaner and easier to maintain, but not backward compatible (we can leave the old names defined in order not to break compilation, but the behavior will definitely change). 

2 is backward compatible, but the macro will be a PITA to do correctly and a lot of work for scavenging where to put the change... 

Any suggestion? Hint? Brilliant idea?

Rmano, 2019-06-06:
Thinking about it, whichever the way forward `\ctikzscalediodes` above will be easy to maintain; maybe I should put it into the next release...

asinghvi17, 2019-06-06:
About the break in backward compatibility - since the older versions are now bundled with CircuiTikZ, this shouldn't be too much of an issue...

Rmano, 2019-06-06:
> About the break in backward compatibility - since the older versions are now bundled with CircuiTikZ, this shouldn't be too much of an issue...

For LaTeX it's true. For ConTeXt I really don't know --- I know zilch of it ;-).

Rmano, 2019-06-08:
@mredaelli @sistlind @der-stefan, what do you think about this issue?

Rmano, 2019-06-23:
After much thinking... the data is: 
- we have more than 850 (!) parameters in `pgfcirc.define.tex`;
- it would be nice to be able to change the relative size of categories of components all together
- the thing must be maintainable in the future.

So my idea now is to (mildly) lose backward compatibility *for the parameters not described in the manual*. I can define several classes of components; I am thinking, for example: 

> grounds, resistors, inductors, capacitors, diodes, sources, instruments, transistors, tubes, amplifiers, logic gates, chips

 
and define for each of them a parameter, like 

    \ctikzset{resistors/scale=1.0}

that will control in a simmetric way height and width. Next, I will collapse all the common parameters to just the two (or three) `width`, `height`, (`depth`) to one per category (as it is nicely done for the `tubes` now). All the other, more specific, parameters should be proportional to the three sizes above, and will be described in the manual. 

Very specific, one-off, size change (like a big, high-power transistor or resistor) can be dealt with styles.

All the evicted parameters will be set to give a warning (see [here how](https://tex.stackexchange.com/questions/495828/retrieve-the-name-of-the-key-in-a-pgfkeys-code-macro)), at least during a couple of releases.

That would be quite a bit of work, to be done slowly, so I tink I will try to do a branch to do it... what do you think?

Rmano, 2019-06-24:
This will be **much** more complex and slow than I suspected. It will need quite a complex review... I hoped that simply defining a kind of `\ctikzdimof{family}{key}` would suffice, but I stopped on the fact that the expansion rule of TeX are much, much more complex than simple function definition. Sigh.

afcuttin, 2019-07-02:
So, for the time being, which is the best approach to follow?

Rmano, 2019-07-02:
> So, for the time being, which is the best approach to follow?

Difficult to say. It will quite impossible to maintain backward compatibility, but when it will happen, a will release also a "frozen" `circuitikz-x.x.x.sty` to deal with old circuits. What I am doing now is looking at `pgfcirc.defines.tex` and change `width` and `height` of the component I want different...

Rmano, 2019-07-13:
I am working on a backward compatible idea as a first step (and then the simplification will follow, if I have the time and will...). I am publishing this here to see if somebody has suggestions or whatever. Notice that it is important to maintain the change of scale in `saveddim` and `savedanchors` to avoid anchors mismatches when changing the scale locally.

You can check the idea on the branch `scaling` on my clone: https://github.com/Rmano/circuitikz/tree/scaling 

The idea is shown in the following example: 

```latex
\documentclass[border=10pt, multi=page]{standalone}
\usepackage[T1]{fontenc}
\usepackage[siunitx, RPvoltages]{circuitikz}

\usepackage{ctikzmanutils}
\newcommand{\anchorsofnode}[2]{%
    \foreach \n/\a/\d in {#2} \path(#1.\n) \showcoordb(\n)<\a:\d>;
}

\newcommand{\showbipole}[3][]{%
    \begin{circuitikz}[#1]
        \draw (0,0) to[#2, name=r1] ++(3,0) coordinate(a);
        \ctikzset{#3/scale=0.5}
        \draw (a)  to[#2, name=r2] ++(3,0) coordinate(a);
        \ctikzset{#3/scale=1.5}
        \draw (a)  to[#2, name=r3] ++(3,0) coordinate(a);
        \anchorsofnode{r1}{nw/145/0.3, ne/45/0.3, se/-45/0.3, sw/-145/0.3}
        \anchorsofnode{r2}{nw/145/0.3, ne/45/0.3, se/-45/0.3, sw/-145/0.3}
        \anchorsofnode{r3}{nw/145/0.3, ne/45/0.3, se/-45/0.3, sw/-145/0.3}
    \end{circuitikz}\par\medskip\par%
}

\begin{document}
\vspace*{1cm}
    \showbipole{R}{resistors}
    \showbipole[european]{R}{resistors}
    \showbipole{vR}{resistors}
    \showbipole[european]{vR}{resistors}
    \showbipole{phR}{resistors}
    \showbipole[european]{phR}{resistors}
\end{document}
```

which for now is like that: 

[image]

Rmano, 2019-07-15:
I pushed more changes on that branch. Look: 

[image]

Rmano, 2019-07-28:
PR #255 is ready to review. It's quite big... but my (evil?) plans do not stop here. I will try to add global styling of fill color and thickness on a per-class basis, so that you can have a kind of style-sheet for your circuits.

Rmano, 2019-08-04:
Merged #261; closing issue!

closed 2019-08-04 by Rmano