circuitikz/circuitikz #219: Make dcisource more discernable

YanWenger, 2019-06-07:
Dear circuitikz developers,

we regularly use the dcisource component at my workplace and have the issue that it is in some cases difficult to distinguish from other circuit elements (we use the component american current source as a *voltage* source).

Would it be possible to increase the white space in the circle as in the example labeled "J" below to improve the visibilty of the component?

[image]

Rmano, 2019-06-08:
Ok, well seen. I did the angle of the broken circle configurable. Notice that `angle` must be between 0 and 90, otherwise havoc will arise ;-).

One question though: I never used these symbols; is the `dcvsource` correct or should be open too?  

```latex
\begin{circuitikz}[
    ]
    \draw (0,0) to[dcvsource] ++(2,0) to [dcisource, fill=yellow] ++(2,0) ;
    \ctikzset{bipoles/dcisource/angle=60} 
    \draw (0,-1) to[dcvsource] ++(2,0) to [dcisource, fill=yellow] ++(2,0) ;
    \ctikzset{bipoles/dcisource/angle=30} 
    \draw (0,-2) to[dcvsource] ++(2,0) to [dcisource, fill=yellow] ++(2,0) ;
\end{circuitikz}
```

[image]

This will be in the next release of `circuitikz`, but if you want, you can use it right now you can just copy the following blurb into your document preamble (*after* loading `circuitikz`):

```latex
\makeatletter
\ctikzset{bipoles/dcisource/angle/.initial=80}
%% DC Current Source with open shape
\pgfcircdeclarebipole
{}
{\ctikzvalof{bipoles/dcisource/height}}
{dcisource}
{\ctikzvalof{bipoles/dcisource/height}}
{\ctikzvalof{bipoles/dcisource/width}}
{
    \pgfsetlinewidth{\pgfkeysvalueof{/tikz/circuitikz/bipoles/thickness}\pgfstartlinewidth}
    \pgfpathmoveto{\pgfpointorigin}
    \pgfpathellipse{\pgfpointorigin}{\pgfpoint{0}{\pgf@circ@res@up}}{\pgfpoint{\pgf@circ@res@left}{0}}
    \pgf@circ@maybefill
    \edef\@@angle{\ctikzvalof{bipoles/dcisource/angle}}
    \pgfpathmoveto{\pgfpointpolar{\@@angle}{\pgf@circ@res@up}}
    \pgfpatharc{\@@angle}{-\@@angle}{\pgf@circ@res@up}
    \pgfpathmoveto{\pgfpointpolar{180-\@@angle}{\pgf@circ@res@up}}
    \pgfpatharc{180-\@@angle}{180+\@@angle}{\pgf@circ@res@up}
    \pgfusepath{draw}

    \pgfpathmoveto{\pgfpoint{.7\pgf@circ@res@left}{\pgf@circ@res@zero}}
    \pgfpathlineto{\pgfpoint{.6\pgf@circ@res@right}{\pgf@circ@res@zero}}
    \pgfusepath{draw}
    \pgfscope
        \pgftransformshift{\pgfpoint{.5\pgf@circ@res@right}{\pgf@circ@res@zero}}
        \pgfnode{currarrow}{center}{}{}{\pgfusepath{stroke}}
    \endpgfscope
}
\makeatother
```

Rmano, 2019-06-08:
I will merge #220  and close this now; if you feel that the `dcvsource`should be equally open, please tell me or point me to any source about the "official" shapes... 

Thanks!

closed 2019-06-08 by Rmano

YanWenger, 2019-06-12:
Hello Romano,

 

thank you very much for the quick answer! My colleagues and I are very happy with your proposed solution.

 

I think the dcvsource is correct the way it is. The only “open” source I have ever seen is this current source. 

 

 <https://user-images.githubusercontent.com/6414907/59144744-4f224580-89db-11e9-8a0b-2ffa3739e764.png>

PR #220 (2019-06-08): Add angle parameter to dcisources
See https://github.com/circuitikz/circuitikz/issues/219

Fixes #219