circuitikz/circuitikz #221: Description of triode

jesseopdenbrouw, 2019-06-08:
Hi all,

I would like to commit the description of a triode tube (I couldn't find one in Circuitikz), but I'm not sure if my coding style is correct to the standards. For instance, a triode is a tripole, but I would like to add a pentode in the future which is a five-pole. And maybe a CRT. So is it an option to define a new branch/directory in pgfkeys with ../tube/triode/.. ?

I have created a repository on GitHub [here](https://github.com/jesseopdenbrouw/circuitikz/tree/master/triode).

Greetings from Holland,

Rmano, 2019-06-08:
Very nice component, I like it. Some comment about the code:

1. do not change the `\pgf@circ@Rlen` length (I know it should be limited by scoping, but anyway, it can be quite confusing). At the start of `pgfcirc.defines.tex` we define 
```latex
\newdimen\pgf@circ@res@up
\newdimen\pgf@circ@res@down
\newdimen\pgf@circ@res@zero
\newdimen\pgf@circ@res@left
\newdimen\pgf@circ@res@right
\newdimen\pgf@circ@res@other
\newdimen\pgf@circ@res@step
\newdimen\pgf@circ@res@temp
``` 
as dimensions you can freely use in your drawings. It is quite customary that you use `...up`, `. .down`, `...left` and `...right` as the cardinal points of the square enclosing the component.

2. Please define a global `tripoles/tubes/width` and `height` so that if you add more components we can change the size just in one place (do not repeat an old error... ;-), see #218 )

3. you have to define all the geographic anchors. If `north` is not defined, for example, then positioning like `[..., below,...]` will not work. 

About where to put them in files, feel free to decide. I would put them all together in `pgfcirctripoles`, to keep then near, but I really do not have any strong opinion about that. 

If you can prepare a pull request, better, because your autorship will be maintained; if you are new to `git` I can help (but there is a basic guide in https://github.com/circuitikz/circuitikz/blob/master/CONTRIBUTING.md )

asinghvi17, 2019-06-08:
@jesseopdenbrow, if you want to submit a pull request, then you'll have to rename your current repository.

jesseopdenbrouw, 2019-06-08:
@Rmano : (1) I will update the code not using `\pgf@circ@Rlen`, (2) something line `\ctikzset{tubes/width/.initial=0.20}`? And for the components own paramaters `\ctikzset{tubes/triode/anode distance/.initial=0.20}`? (3) will do that.
@asinghvi17 : done.

Rmano, 2019-06-08:
> @jesseopdenbrouw, if you want to submit a pull request, then you'll have to rename your current repository.

You really should start by cloning the main `circuitikz` repository...

About the code, yes, normally you do 
```latex
\ctikset{tubes/width/.initial=0.5} 
```
and in the code start with a 
```latex
\pgf@cir@res@up = \ctikzvalof{tubes/width}\pgf@circ@Rlen
```
...and use that as the base length.

asinghvi17, 2019-06-08:
Github wouldn't let `circuitikz` be forked if it detects another repo named `circuitikz` on the user - sorry, should have been more specific about that.

Rmano, 2019-06-08:
> Github wouldn't let `circuitikz` be forked if it detects another repo named `circuitikz` on the user - sorry, should have been more specific about that.

Ah, yes, you're right!

jesseopdenbrouw, 2019-06-08:
This triode doesn't have both a height and depth because is it a circle shaped schematic by default (oh well, there are other schematic drawings, but I will save that for the pentode). Also I would like to scale the triode by 1.06066 because then a side of the bounding box is exactly 1.5 and not 1.4142....

Rmano, 2019-06-08:
No problem, use just `width`. And I do not understand really why you have to scale this... you can do a lot of calculation using the `\pgfmathsetmacro` and `\pgfmathsetlenght` macro (see for example the anchors of  the `diamondpole` shape). But whatever you want, if you keep it into a `\pgfscope`...`\endpgfscope` pair.

jesseopdenbrouw, 2019-06-10:
Ok,

I'm almost ready with the tubes: diode (named diodetube to avoid clashes), triode, tetrode and pentode. They are basically the same, only the number of grids differ. I've also created a basic version of the documentation. It can be found [here](https://github.com/jesseopdenbrouw/circuitikz_enhancement/tree/master/tubes).

Now the next thing is to incorporate the code in the main branch. I've created a local copy of the circuitikz/circuitikz repo. I presume I can just add the code and doc to the local repo copy. But then I have to create a pull request and that's a  bit of a mistery to me. So how to proceed and create a pull request.

asinghvi17, 2019-06-10:
@jesseopdenbrouw, you need to fork the `circuitikz` repo through the Github web interface first.  Then, you can clone the fork to your computer, integrate the code, and create a commit.  That workflow would go something like this (in the shell):

```bash
git clone https://github.com/jesseopdenbrouw/circuitikz
cd circuitikz
# edit the code
git commit -am "Added tubes" # or whatever commit message you want
git push
```

Then, you can create a pull request (in the pull requests tab).  Or, if you navigate to the main page of this repo, then it might show that you have recently pushed to the master branch of your fork, and give you the option to make a pull request then and there.

asinghvi17, 2019-06-10:
Also, it seems that your repo is private; I cannot access https://github.com/jesseopdenbrouw/circuitikz_enhancement/tree/master/tubes.

jesseopdenbrouw, 2019-06-10:
Oops, public now.

Rmano, 2019-06-10:
Two comments, the first one esthetical: I really like more the round ones, but maybe this is the most "standard" way. Nice anyway. 

About the PR and the clone; if you do not want to go through it, I can integrate them (and of course I will maintain the attribution into files and manual), but 
1. the repository file will not have the correct autorship;
2. you shoud add a LICENSE file anyway to the repo and to your files... see the main `circuitikz` README.

Rmano, 2019-06-10:
The other thing I do not like much is that any tube has different parameters --- yes, I know, it's like it's done now, but I think it is an error. You should just define a `tube/width` (and similar parameters); I think the user should be able to change all the tubes aspect together. And if you want a specific tube to be distinct, you can always say 
```latex
... node[triode, circuitikz/tube/width=2, ...]{}
```

asinghvi17, 2019-06-11:
That could also be put in the manual, as could a description of a style that would do that for you.

asinghvi17, 2019-06-11:
@Rmano, about the first point - there's a way that you can preserve authorship while doing the integration yourself.  

In a commit's description text, or two newlines down from the end of the description, you can include a `Co-authored-by` field.  It would look something like this:

```
Added tubes (diode tubes, triodes, tetrodes, and pentodes)

This commit is thanks to Jesse op den Brouw, who defined these components.

```

jesseopdenbrouw, 2019-06-11:
@Rmano : I've created a fork of circuitikz and created a local copy on my harddrive, so it wouldn't be a problem to create a pull request. About the heights: i'll make a generic key for the heights of the tube (and the widths), but the stretched versions are far more common in schematics the the round ones. Also, this could be a problem for rendering the pentode as the grids would be too close to the anode and cathode grids or the grid spacing would be the small. I'm planning to create a multigrid tube so the user can, for instance, create a hexode, heptode and octode (but these tubes are very rare). In contrast to your opinion, I like separate parameters more than generic ones. But that's my opinion. I have no problem to create most, if not all, generic parameters for the tubes.

Rmano, 2019-06-11:
@jesseopdenbrouw Ok, nice, I really prefer the PR workflow. 

About the shape: this is an artistic point-of-view, and given that the artist is you, I will accept whatever seems better to you. 

About the sizes. I am thinking, for future release, to change the parameters so that every "group" of components have a kind of scale. For example, right now `diode` will change the size of all the diodes (almost), given that there is little sense in having, for example, the tunnel diode different in size from a Zener diode --- unfortunately this is not so for transistors, generators, etc. That makes styling circuits a pain in the back. As I comment in issue #218, I think that the best and more generic thing would be to rewrite the million of parameters to a reasonable few ones so that I can say "I want generators and diodes smaller, and transistors bigger" in a generic way. But this is quite a big change (booooooring one, too), and not backward compatible. 

Another possibility is having a generic size (like `tubes/width`) and then specific sizes (`tubes/triode/width`) *relative* to the first one --- this is the best of both world, but I do not know if the hassle is worthwhile, given that you can use locality and style to change the odd element...

So again, in this case, is up to you as author, but really someone does want triodes smaller than pentodes in the same diagram _always_?  

I really would like a bit of feedback from @mredaelli @sistlind and @der-stefan  about this and #218... since it's quite a big decision.

jesseopdenbrouw, 2019-06-11:
Well, I have merged all individual keys to a generic set. and did some code runtime speed up. It should be more or less to the standards.

Rmano, 2019-06-11:
Seems really nice code to me. And if you did *any* speed optimization, is more "up to standard" than mine ;-)!

jesseopdenbrouw, 2019-06-11:
There were some tricky calculations for drawing the grids, e.g. the drawing of the dashed grid lines. I used some dimension registers to do calculations once and the use them multiple times. Also I try to minimize `\pgfmathparse` as much as I can...

Rmano, 2019-06-12:
Perfect, I like it very much. If you prepare a PR, I'll be happy to merge it.

jesseopdenbrouw, 2019-06-12:
I have one question though. If I use `label=...` in the node desscription, the label is rendered at the centerpoint. Is there a way to intercept that? The way I want to use it is now tested with the `model=...` option, whch prints `...` at the right side of the node (see the latest version). Also I want to do some tube outline drawing test when the user sets the width greater than the height. Currently, the outline is malformed.

Rmano, 2019-06-12:
The `label` (`l`, `l_` and company) is really only defined for path-style (bipoles) components, not for nodes. I wouldn't do a lot of work there; in the end, you can always say 

```latex
\path node[triode](T1){} (T1.east) node[left]{My label};
```

which integrates well with normal TikZ drawing. 
You can also use the anchor `text` to position the text of the node argument --- you can use a bit of math to calculate where to but that text (see the `op amp` component for example), and you'll be able to say

```latex
\path node[triode](T1){My label};
```

Rmano, 2019-06-12:
A thought: maybe to have the `label` (which is a standard TikZ feature) you need the `\anchorborder` defined... which is quite a PITA... see for example the `diamonpole` one. The anchor is described in the manual of TikZ, 101.5.3 (page 1037 in my copy).

Rmano, 2019-06-12:
Also, I have a bit of fear that the `model` keyword is too common and can create clashes with other packages... what about using `tube model`?

jesseopdenbrouw, 2019-06-12:
`tube model` it is... I like that better than fooling around with `label`. Also, `text` could be an equivalent, but I have to figure that out.

Rmano, 2019-06-12:
The `text` thing is quite easy. In the anchor, you have to specify the bottom left corner; you have the text in a box when calling the anchor. So for example: 
```latex
\anchor{text}{
     \pgfpoint{-.5\wd\pgfnodeparttextbox}
                    {\dimexpr.5\dp\pgfnodeparttextbox-.5\ht\pgfnodeparttextbox}
}
```
you put the anchor one half of the width to the left, and one half of the total heigth down, resulting in a text with its center at (0pt,0pt).

jesseopdenbrouw, 2019-06-12:
Ok, implemented the `text` anchor, but there's another problem, when using xscale=-1. Then the text is also scaled....

Rmano, 2019-06-12:
> Ok, implemented the `text` anchor, but there's another problem, when using xscale=-1. Then the text is also scaled....

Yes, it's known. It's a basic problem --- we can't manipulate the internal text. It's noted in the `circuititikz` manual too; look at section 3.2.1. You have to use 
```latex
node[rground, yscale=-1] {\scalebox{1}[-1]{ GND}}
```

jesseopdenbrouw, 2019-06-12:
Yes, you're right. I'll remove the `tube model` option and `text` anchor.

jesseopdenbrouw, 2019-06-13:
Ok, ready now. I will put the tubes' description at the end of the tripole file, right after the magnetron, and put the keys in the pgfcirc.defines.tex file. Ok?

Rmano, 2019-06-13:
> Ok, ready now. I will put the tubes' description at the end of the tripole file, right after the magnetron, and put the keys in the pgfcirc.defines.tex file. Ok? And the documentation right after the magnetron.

Yes! And the documentation together with the magnetron --- I would put your tubes *before* the magnetron, they're more generic.

jesseopdenbrouw, 2019-06-13:
OK, will do that and the create a pull request.

jesseopdenbrouw, 2019-06-13:
Created pull request -;)

Rmano, 2019-06-16:
Tubes (and nice ones!) are in. Thanks a lot, @jesseopdenbrouw !

closed 2019-06-16 by Rmano

jesseopdenbrouw, 2019-06-16:
I had such a fun coding them... ;-)

Rmano, 2019-06-16:
Do not worry, there will be *a lot* of fun ahead with #218 and (if I ever find what happens) #211... happy to see you contribute!