circuitikz/circuitikz #226: Circuitikz nodes coordinates problem

BobUd, 2019-06-10:
Hello everybody and many thanks for your work.
I'm trying to draw a simple network made of resistors, however compilation fails for a given resistor and no useful (for me!) message is given. It seems to be a problem of nodes coordinates. Hereafter are the error message and  a minimal version of the LaTex input file that gives the error. To check if it was a problem of FORMAT of nodes coordinates, (line 13 of the file), I also tested another format (line 15), but the error remains. If I change the COORDINATES VALUES  the problem disappears.
My circuitikz version is version 0.3.0. I use TexShop for editing files.
Thanks again

BOB
----------------------------
This is the error message: 

./GB4_3_not_working.tex:13: Undefined control sequence.
\tikz@anchor ...@circ@bipole@voltage@label@anchor 
                                                  
l.13 ...or=black]     ( 4.50175E+00,  7.78710E-01)
                                                   node [above] {24};
? r
OK, entering \nonstopmode...

----------------------------
and this is the input file:

\documentclass[11pt]{standalone}
\usepackage{tikz}
\usepackage{circuitikz}
\usepackage{epstopdf}
%
\newcommand{\tikzScale}{}
\begin{document}
\renewcommand{\tikzScale}{1.0}
%--------------------------------------------------
\begin{circuitikz} [ american voltages , scale=3]
\ctikzset{label/align = straight}
% This is the original instruction, not working:
\draw[] ( 4.23910E+00,  1.99187E+00)     node [above] {6}     to [american resistor=G34 , i=L34, v=$u$ , o-o, color=black]     ( 4.50175E+00,  7.78710E-01) node [above] {24};
% This is a modified instruction, with exponential format switched to fixed, still not working:
%\draw[] ( 4.23910,  1.99187)     node [above] {6}     to [american resistor=G34 , i=L34, v=$u$ , o-o, color=black]     ( 4.50175,  0.778710) node [above] {24};
\end{circuitikz}
%--------------------------------------------------
\end{document}

Rmano, 2019-06-10:
Which version are you using? 0.8.3 had a bug with straight labels...

Rmano, 2019-06-10:
Works for me on 0.9.0:
[image]

BobUd, 2019-06-10:
Dear Rmano,
I downloaded CTAN new version of Circuitikz 0.9.0, copied circuitikz directory in my local TexLive distribution:

usr/local/texlive/2017/texmf-dist/tex/generic/

compilation starts well, however it stops when the following packages are going to be included by the compiler (corresponding to  line 344 and following of circuitikz.sty):
pgfcircmultipoles.tex
pgfcircflow.tex

if give the run-time command "ignore all", in the end the figure you obtained is produced.
Where am I wrong?

This is part of the .log message:
-----------------------------------------
(/usr/local/texlive/2017/texmf-dist/tex/generic/circuitikz/pgfcirc.defines.tex
\pgf@circ@res@up=\dimen162
\pgf@circ@res@down=\dimen163
\pgf@circ@res@zero=\dimen164
\pgf@circ@res@left=\dimen165
\pgf@circ@res@right=\dimen166
\pgf@circ@res@other=\dimen167
\pgf@circ@res@step=\dimen168
\pgf@circ@res@temp=\dimen169
\pgfstartlinewidth=\dimen170
\pgf@circ@Rlen=\dimen171
) (/usr/local/texlive/2017/texmf-dist/tex/generic/circuitikz/pgfcircutils.tex)
(/usr/local/texlive/2017/texmf-dist/tex/generic/circuitikz/pgfcircshapes.tex)
(/usr/local/texlive/2017/texmf-dist/tex/generic/circuitikz/pgfcircmonopoles.tex
) (/usr/local/texlive/2017/texmf-dist/tex/generic/circuitikz/pgfcircbipoles.tex
)
(/usr/local/texlive/2017/texmf-dist/tex/generic/circuitikz/pgfcirctripoles.tex)

(/usr/local/texlive/2017/texmf-dist/tex/generic/circuitikz/pgfcircquadpoles.tex
)
/usr/local/texlive/2017/texmf-dist/tex/latex/circuitikz/circuitikz.sty:344: I c
an't find file `pgfcircmultipoles.tex'.
l.344 \input pgfcircmultipoles.tex
                                  
(Press Enter to retry, or Control-D to exit)
Please type another input file name: s

-----------------------------------------

Thank you very much for your prompt answers!

Bob

Rmano, 2019-06-10:
If you have a Unix TeXlive distribution (I do not know about other distribution or operating systems), the zip (circuitikz.tds.zip) must be unzipped in the main directory of the distribution (in your case, that would be `/usr/local/texlive/2017/texmf-dist/`, not in a subdirectory. And if it's in a system directory, you have to rebuild the database (with `mktexlsr` or something like that, with or without `sudo` depending on your configuration). 

Otherwise, you can create the directory `texmf` under your home, and unizp the file there --- that should not need any indexing and it should override your distro file.

Anyway, the fastest way to use the last version  of the package is: go to `https://circuitikz.github.io/circuitikz/`, download the `circuitikzgit.sty` file from there, put it in the same dir as your main file, and change `\usepackage[...]{circuitikz}` with `\usepackage[...]{circuitikzgit}`.

BobUd, 2019-06-13:
I successfully installed the new version of package Circuitikz. O this purpose hat to rebuild the LaTex filename database using the command "texhash".
Everything seems to be ok.
Many thanks for the excellent assistance!
BobUd

closed 2019-06-13 by BobUd