circuitikz/circuitikz #229: Need to do a 0.9.2 shortly

Rmano, 2019-06-17:
Ok, I realized that the ConTeXt side has been in bad shape since 0.9.0. Unfortunately, I do not know it and it seems that no one is checking it...  I need to: 

- [ ] add manually the new includes in `t-circuitikz.tex` (I thought it was automatic!) 
- [ ] remove the mention of old versions in README (it works only in LaTeX --- unless someone can help me generate the files, they're not it). 

Too late to target 0.9.1, it will need a 0.9.2 shortly on.

sistlind, 2019-06-17:
the 'git'-versions are also generated for context at the gh-pages branch,
but to be honest, I never checked them.

If a filename/the defines have changed, it has to be adjusted manually.
Maybe the same is true if the paket option changes, but here I am not sure.

Rmano, 2019-06-17:
Steps for 0.9.2
- [ ] freeze the repo (just a bit) 

When ready, prepare a PR that: 

- [ ] set version number and date in circtuitiks.sty, t-circuitikz.sty, manual and changelog
- [ ] commit everything
- [ ] run "make ctan" this will produce a zipfile. check the contents of this file (especially the version & date of docs & package)
- [ ] go to page https://ctan.org/pkg/circuitikz
- [ ] click on upload(at the top)
- [ ] upload the zipfile and fill the missing fields ( I copied the changelog to the announcement field)

Now the repository is freezed; I'll wait for:

- [ ] Confirmation of the correct upload 
- [ ] If the package is ok it will take a few days to spawn over all servers

And after that:

- [ ] Push your commit to github
- [ ] Add a github release with same infos & zip file.
- [ ] Add at the the link to the circuitikz-git file of the new release to the circuitikz github pages site (edit https://github.com/circuitikz/circuitikz/blob/gh-pages/index.html manually )
[image]

- [ ] reinsert the git headline in the changelog file.

...and now development will continue.

Rmano, 2019-06-17:
> If a filename/the defines have changed, it has to be adjusted manually. Maybe the same is true if the paket option changes, but here I am not sure.

...we need someone using it in ConTeXt giving feedback...

Rmano, 2019-06-21:
Started...

Rmano, 2019-06-21:
Trying to upload, I have an internal server error on CTAN. I reported the problem, will try again later.

Rmano, 2019-06-22:
> Trying to upload, I have an internal server error on CTAN. I reported the problem, will try again later.

Still kaputt...

Rmano, 2019-06-22:
Ok, ctan has come back to life. Uploaded!

jesseopdenbrouw, 2019-06-22:
I am trying to compile this short TeX file with ConText, It produces the resistor, I will try more.

closed 2019-06-23 by Rmano