circuitikz/circuitikz #234: Electronic tubes not working in ConTeXt

jesseopdenbrouw, 2019-06-24:
Rendering the electronics tubes (except magentron) fails when using ConTeXt. This is due to ConTeXt not defining `\@tempcnta`. I am working on a fix.

Rmano, 2019-06-24:
Yep, nicely spotted. I have defined in `pgfcircmultipoles.tex` this:
```latex
% counters for pin accounting...
\newcount\pgf@circ@count@c
\newcount\pgf@circ@count@b
```
to add to the `\pgf@counta` used somewhere in `tikz`. Maybe we can add them to the start of `pgfcirc.defines.tex` where the various temporary lengths used in shapes are defined.

Rmano, 2019-06-24:
Well, worth fixing, but I really do not know how many users using ConTeXt are here, so no need to hurry...

jesseopdenbrouw, 2019-06-24:
Well, as a quick workaround, include the following lines:

```
%% Insert these four lines to prevent TikZ from failing to compile
\unprotect
\def\XC@tgt@mod#1{#1}
\def\XC@sdef#1#2{\edef#1{#2}}
\protect

% Define use counter and dimension
\catcode`\@11\relax
\newcount\@tempcnta
\newdimen\@tempdimc
\catcode`\@12\relax

```

jesseopdenbrouw, 2019-06-24:
> Well, worth fixing, but I really do not know how many users using circuitikz in ConTeXt are here, so no need to hurry...

Well, you asked for ConTeXt testers ;-)

jesseopdenbrouw, 2019-06-24:
> Yep, nicely spotted. I have defined in `pgfcircmultipoles.tex` this:
> 
> ```
> % counters for pin accounting...
> \newcount\pgf@circ@count@c
> \newcount\pgf@circ@count@b
> ```
> 
> to add to the `\pgf@counta` used somewhere in `tikz`. Maybe we can add them to the start of `pgfcirc.defines.tex` where the various temporary lengths used in shapes are defined.

Also note that TikZ defines `\c@pgf@counta` {b,c,d}. So I think I'll use those.

Rmano, 2019-07-01:
> Also note that TikZ defines `\c@pgf@counta` {b,c,d}. So I think I'll use those.

Ah --- nice find. Yes, that would be great.

Rmano, 2019-07-02:
Using `\c@pgf@counta` created bug #243 --- please look at it. I am not sure how it could happen, but it did...

commit 160b66d8 (2019-07-04): Update electronic tubes (#247)

* Fix tubes for ConTeXt
* Fixes #234 

* changed anchor grid to anchor control

* changed anchor grid t
closed 2019-07-04 by Rmano