circuitikz/circuitikz #235: Key style 'voltage dir' does not affect current direction when used for single component

FloMiLe, 2019-06-26:
I do not know whether this is intended behavior. But I noticed that the key style 'voltage dir' behaves differently when set as an environment parameter rather than a component parameter.

Voltage and current direction are affected:

    \begin{tikzpicture}[ american, voltage dir=RP ] 
    \draw (0,0) to[V, *-o, v=$v_1$, i=$i_1$, ] (2.5,0); 
    \end{tikzpicture}

Only voltage direction is affected:

    \begin{tikzpicture}[ american ] 
    \draw (0,0) to[V, *-o, v=$v_1$, i=$i_1$, voltage dir=RP] (2.5,0); 
    \end{tikzpicture}

For me, it would be more consistent and desirable to also have the effect of the first MWE in the second MWE.

Rmano, 2019-06-27:
Well --- it's not exactly expected, so it's not a feature, but... Let's see what happens: 

```latex
\documentclass{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[nooldvoltagedirection]{circuitikz}
\begin{document}
Default:

\begin{tikzpicture}[ american]
\draw (0,0) to[V, *-o, v=$v_1$, i=$i_1$, ] (2.5,0);
\end{tikzpicture}

\bigskip
In environment:

\begin{tikzpicture}[ american, voltage dir=RP ]
\draw (0,0) to[V, *-o, v=$v_1$, i=$i_1$, ] (2.5,0);
\end{tikzpicture}

\bigskip
In component:

\begin{tikzpicture}[ american ]
\draw (0,0) to[V, *-o, v=$v_1$, i=$i_1$, voltage dir=RP] (2.5,0);
\end{tikzpicture}

\bigskip
In draw:

\begin{tikzpicture}[ american ]
    \draw[voltage dir=RP] (0,0) to[V, *-o, v=$v_1$, i=$i_1$] (2.5,0);
\end{tikzpicture}
\end{document}
```

will produce: 

[image]

As you can see, the voltage direction is drawn when the component is drawn (it's a `behindbackground` path), so it is using the localized change of `voltage dir`, which is contained in the component TeX environment by the grouping scope. The current, though, add a node to the path (the small arrow with its label), and nodes in TikZ are added to the path *at the end of the path drawing*, so it is using whatever the defaults for the enclosing `draw` or `path` scope are. 

Basically, the `voltage dir` is not supposed to be used at component level. To change voltage and current direction at the component level, you should use `v=`, `v<=`, `v<^=` and the corresponding `i` with `<>_^` styles. 

The key is provided as a transition tool because it makes easy to re-use circuits drawn for older versions. So if you are using the new `RPvoltages` global option, and you copy a circuit from, say, v0.4, you can just add `voltage dir=old` (or `voltage dir=noold` if you are using a v0.5 to v0.8.3) and you do not need to check every voltage in it. 

So --- this is quite impossible to fix; it's just not coding, it's that if I write

```latex
\draw (0,0) to[V, *-o, v=$v_1$, i=$i_1$, voltage dir=RP] (2.5,0) 
            to[V, voltage dir=old]  ++(2,5,0);
```

when the current arrows are drawn, at the end of the path, there is no way to choose one or the other style.

Rmano, 2019-06-27:
In PR #236  I have added a notice in the manual to clarify usage. Thanks for raising the issue.

Rmano, 2019-06-28:
I am closing it. Please feel free to reopen if needed.

closed 2019-06-28 by Rmano

FloMiLe, 2019-07-21:
I meant to say thanks for you detailed reply...

Rmano, 2019-07-21:
> I meant to say thanks for you detailed reply...

You're welcome!