circuitikz/circuitikz #238: PMOS arrow symbol

Arkh42, 2019-07-01:
Dear developers,

Currently, the __arrowmos__ option draws correctly the arrow symbols depending on the type of MOS transistor. However, the full circle is still drawn on the PMOS symbol, while it shouldn't be.
I have read many articles and books in microelectronics (this is my field of work) and I've never see that. Either there is an arrow or a circle, as shown [here](https://en.wikipedia.org/wiki/MOSFET#Circuit_symbols) and [there](https://i.stack.imgur.com/1gpMy.jpg).
Could you please do something about it?
If it can be of any help, I found [this](https://tex.stackexchange.com/questions/237682/tikz-change-symbol-of-pmos).

Thank you very much for your great job!

Rmano, 2019-07-01:
I will propose a PR with the ability to do this: 

[image]

Rmano, 2019-07-01:
Although I agree that with `arrowmos` the default should be `nocircle`, I do not like backward-incompatible changes a lot, so I think I will not touch the default. Opinions welcome.

closed 2019-07-02 by Rmano

Arkh42, 2019-07-02:
I understand the problem of backward incompatibility and your desire to not change the default.
Probably not a big deal if there is a note in the manual.

Is the option "emptycircle, noarrows" available as well?

Rmano, 2019-07-02:
> Is the option "emptycircle, noarrows" available as well?

Yes: 
```latex
\ctikzset{tripoles/pmos style/emptycircle, tripoles/mos style/no arrows}
\draw (0,-2) node[info]{emptycircle} node[pmos]{} (2,-2) node[nmos]{}
```

[image]

I added things to the manual; please check it out. Suggestions and changes are welcome!

PR #240 (2019-07-02): Reorganized the option for dot/arrow of MOSes.
Now you can have a PMOS with arrow and without dot.

Fixes #238