circuitikz/circuitikz #239: Moving coupler2 in a place different than (0,0) leaves pieces around

afcuttin, 2019-07-01:
If I move `coupler2` in a place different than (0,0), some pieces are left around (arrows and label)
In the following code, node `A` is placed in (0,0), and `coupler2` in (2,2).
The pieces around node `A` do not move according to `coupler2` node definition.

Everything works properly if I use `coupler` node, instead.

[fig-fat.pdf](https://github.com/circuitikz/circuitikz/files/3345981/fig-fat.pdf)

```
\documentclass[border=2cm,a4paper]{standalone}
\usepackage{siunitx}
\usepackage{pgfplots}
\pgfplotsset{
        compat=newest,
    }
\usepackage{circuitikz}

\def\coord(#1){coordinate(#1)}
\def\coord(#1){node[circle, red, draw, inner sep=1pt,pin={[red, overlay, inner
sep=0.5pt, font=\tiny, pin distance=0.1cm, pin edge={red, overlay
,}]45:#1}](#1){}}

\begin{document}
\begin{circuitikz}
	\node (a) at (0,0) [red] {A};
	\draw
	(2,2) node[coupler2](c1) {\SI{3}{dB}};
	\draw
	(c1.1) \coord(1) % to[short,-o] ++(-0.5,0) \coord(1)
	(c1.2) \coord(2) % to[short,-o] ++(0.5,0) \coord(2)
	(c1.3) \coord(3) % to[short,-o] ++(0.5,0) \coord(3)
	(c1.4) \coord(4) % to[short,-o] ++(-0.5,0) \coord(4)
	;
\end{circuitikz}
\end{document}

```

Rmano, 2019-07-01:
This is quite scary. The problem seems to be that the arc + arrow operation messes with the origin of the shape coordinates, and I really do not know why. I made a pull request that should fix this for you, but if anybody has any idea on what's really happening, please, tell me.

afcuttin, 2019-07-02:
Thanks for handling it right away! 
For the time being, I'm using `coupler`

Rmano, 2019-07-02:
I will merge the fix during this morning, as soon as possible... It will be in the got version soon!

closed 2019-07-02 by Rmano

afcuttin, 2019-07-02:
Thanks!
Just out of curiosity, when will this update be propagated to CTAN (and then, available with the TeXLive manager)?

Rmano, 2019-07-02:
I do not know exactly. I will wait for the fix to #234 (the author is working on it), and given the last days experience, I'll let you find another couple of bugs ;-) (BTW, thanks a lot). I will surely do a release before my vacations, that is, around the end of July.

afcuttin, 2019-07-04:
> I'll let you find another couple of bugs ;-) (BTW, thanks a lot). 

Happy to help! The package is really nice! IT Takes a bit to learn to use it, but is really nice and promising.

PR #241 (2019-07-02): Fix problem with coupler2
It seems that the arc+arrows mingle with the origin of
the coordinates. Insulate them with \pgfscope.

Fixes #239