circuitikz/circuitikz #242: Package circuitikz Warning: You did not specify one of the voltage directions: ...

khaaba, 2019-07-02:
Why do I lend up here that 

> Package circuitikz Warning: You did not specify one of the voltage directions: oldvoltagedirections, nooldvoltagedirections, RPvoltages or EFvoltages Default directions may have changed, please check the manual.

I didn't face this issue earlier when I ran the same file many times earlier. When I checked the package, it's here:

> \ifpgf@circ@explicitvdir\else
>     \PackageWarningNoLine{circuitikz}{%
>         You did not specify one of the voltage directions:\MessageBreak
>         \space\space    oldvoltagedirection, nooldvoltagedirection, \MessageBreak
>         \space\space    RPvoltages or EFvoltages \MessageBreak
>         Default directions may have changed, \MessageBreak
>         please check the manual%
>     }
> \fi

Any clue, I am still learning and could not solve this?

Rmano, 2019-07-02:
> Any clue, I am still learning and could not solve this?

Have you checked the manual? It's complex, and you have to make a decision, so you have to read about it. 
After that, just add an option of the four to the circuitikz package load

    \usepackage[RPvoltages]{circuitikz}

khaaba, 2019-07-02:
I didn't read the updated manual. However, this was never the case earlier. Yes, your optional parameter resolved the warning. Thank you.

closed 2019-07-02 by khaaba