circuitikz/circuitikz #243: Contact nodes missing in rotary switch - parts not displayed

afcuttin, 2019-07-02:
I have tried to use the newly introduced rotary switches, but I am unable to get a "hello world" sample of code to work.

Here is the minimum (not) working example and [its output](https://github.com/circuitikz/circuitikz/files/3349508/rot-swi-mwe.pdf).

```
\documentclass[border=.5cm]{standalone}

\usepackage[RPvoltages]{circuitikz}

\begin{document}
	% \begin{circuitikz} % line 6
		% \node (a) at (0,0) [] {};
		% \node[spdt,name=rotswi] {}; % line 8
		% \draw ($(rotswi)!.59!(rotswi.east)$) to[short,o-] (rotswi.east);
	% \end{circuitikz} % line 10

	\begin{circuitikz}
	    \draw
	    (0,0) -- ++(1,0)
	    node[rotary switch <->=3 in 45 wiper 22, anchor=in](R){}
	    (R.out 1) -- ++(0.5,0)
	    (R.out 2) -- ++(0.5,0)
	    (R.out 3) -- ++(0.5,0);
	\end{circuitikz}

	% \begin{circuitikz}
	%     \draw
	%     (0,0) -- ++(1,0)
	%     node[rotary switch <->=5 in 45 wiper 22, anchor=in](R){}
	%     (R.out 1) -- ++(0.5,0)
	%     (R.out 2) -- ++(0.5,0)
	%     (R.out 3) -- ++(0.5,0);
	% \end{circuitikz}
\end{document}
```
As you can see, the terminations are not printed correctly.

Funny thing is that, fiddling with the code, it happens that uncommenting lines 6, 8 and 10, the ouptut [is the following](https://github.com/circuitikz/circuitikz/files/3349519/rot-swi-mwe.pdf), showing that the rotary switch can indeed behave properly.

The same behavior occurs with the sample code provided [here](https://tex.stackexchange.com/a/493181/177).

In summary, it seems that something (another switch?) has to be called before calling the rotary switch, in order to have it working properly.
Try, for example to comment the first circuitikz environment and uncomment the third (that is, having  active the second and the third circuitikz environment). I obtain the [following output](https://github.com/circuitikz/circuitikz/files/3349539/rot-swi-mwe.pdf) (which also misses some terminals).

Rmano, 2019-07-02:
Which version are you using? I have this: 

```latex
\documentclass[border=.5cm]{standalone}
\usepackage[RPvoltages]{circuitikz}

\begin{document}
	\begin{circuitikz}
	    \draw
            (0,0) node[left]{\pgfcircversion} -- ++(1,0)
	    node[rotary switch <->=3 in 45 wiper 22, anchor=in](R){}
	    (R.out 1) -- ++(0.5,0)
	    (R.out 2) -- ++(0.5,0)  node[right]{\pgfversion}
	    (R.out 3) -- ++(0.5,0);
	\end{circuitikz}
\end{document}
```
Showing: 

[image]

also with the (freshly uploaded( git version: 

[image]

afcuttin, 2019-07-02:
Updated this morning to 0.9.2

Rmano, 2019-07-02:
I also tried your last version, and I have: 

[image]

Can you test with the `circuitikzgit.sty` from the gituhub page? https://circuitikz.github.io/circuitikz/ --- just download the  `circuitikzgit.sty` file, put it into the current directory, and change the usepackage to point to it. Which is your TikZ version? (

afcuttin, 2019-07-02:
Running your MWE with my installed TeXLive returns:
[image]
Running your MWE with `\usepackage{circuitikzgit}` returns:
[image]

Rmano, 2019-07-02:
This is worrying, because you have a newer TikZ than mine...  I will try to see what happens. Circuitikz touches a lot of internals, and this can be a problem. 
I'll try to look into it.

Rmano, 2019-07-02:
Ok, I can reproduce it with PGF 3.1.3. That will be tough...

afcuttin, 2019-07-02:
Is `\pgfplotsset{compat=<something>}` a possible workaround?

Rmano, 2019-07-02:
I hope not... `pgfplots` should have nothing to do with `circuitikz`, unless there is  a big error on my side....

Rmano, 2019-07-02:
I found it. Now I have to think on how to solve it properly. Stay tuned,

Rmano, 2019-07-02:
Basically, this loop in `pgfcircmultipoles.tex`, around line 620:
```latex
 % \typeout{CHANNELS\space\channels\space ANGLE\space\angle}
        \c@pgf@counta=\channels\relax
        \pgfmathsetmacro{\currenta}{-\angle}
        \pgfmathloop%
        \ifnum\c@pgf@counta>0
            % \typeout{LOOP\space\the\c@pgf@counta\space CURRENTA\space\currenta\space RIGHT\space\the\pgf@circ@res@right}
            \pgfscope
                \pgftransformshift{\pgfpointadd{\pgfpoint{\pgf@circ@res@left}{0pt}}{\pgfpointpolar{\currenta}{2\pgf@circ@res@right}}}
                \pgfnode{\cshape}{center}{}{\thisshape-out \the\c@pgf@counta}{\pgfusepath{stroke}}
            \endpgfscope
            \pgfmathsetmacro{\currenta}{\currenta+\stepa}
            \advance` by -1\relax%
        \repeatpgfmathloop
```

was failing, because sometimes it was as if the counter ` \c@pgf@counta` was modified in some of the inner  function. @jesseopdenbrouw --- this may be relevant for issue #234. 

What I am doing is avoiding using that temporary counter anywhere, and just using "ours" own defined.

jesseopdenbrouw, 2019-07-02:
I will look into that. I see that you merged three new counters within CircuiTikZ. I will use them.

Rmano, 2019-07-02:
I just merged #244. In a few minutes you should see the new `circuitikzgit.sty` in the github.io page (I *think* it should be version 1b3b54c). Please test...

jesseopdenbrouw, 2019-07-02:
So it is not in the circuitikz/circuitikz branch?

Rmano, 2019-07-02:
> So it is not in the circuitikz/circuitikz branch?

Yes, it's also in circuitikz/master branch.

Rmano, 2019-07-03:
@afcuttin, did the patched version works for you? ...I'd like to close this.

afcuttin, 2019-07-04:
hi!
Yesterday I was out of office and couldn't test it out.
Will do that shortly, as I'm back on that diagrams.

I'll download the circuitikzgit.sty file and try it out. Is that ok?

Rmano, 2019-07-04:
> I'll download the circuitikzgit.sty file and try it out. Is that ok?

Yes, perfect!

afcuttin, 2019-07-04:
Looks ok now!
[image]
The example in the documentation compiles fine, too
[image]

closed 2019-07-04 by Rmano