circuitikz/circuitikz #245: Transistor bipoles versions

JulianSchneider, 2019-07-03:
With my question [here](https://tex.stackexchange.com/questions/498451/circuitikz-nfet-solid-line/498493#498493), that I have to use nodes for MOSFETs , which makes using a lot of them in a schematic unwieldy. If there were a bipole MOSFETs (like the switch), it would make the diagram a lot cleaner. There is quite some cases where MOSFETs are drawn without the Gate connected to show the direction of the body diode.

Rmano, 2019-07-03:
(Copy of the answer in tex.stackexchange.com).

You can use the "path" style for transistors (look at section 3.18.2 in new `circuitikz` manual). Unfortunately, labels for transistors are buggy (it's in the manual, but with the `bodydiode` are even more buggy), so you have to add them by hand. 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
%\usepackage[siunitx, RPvoltages]{circuitikzgit}
\begin{document}
\begin{circuitikz}[
    ]
        \draw (0,0)
        to [Tnigfete, nogate, bodydiode, invert, name=s1, -*] ++(2,0)
        coordinate(one)
        to [Tnigfete, nogate, bodydiode, invert, name=s2] ++(2,0);
        \draw (s1.center) node[above=0.4cm] {S1}
              (s2.center) node[above=0.4cm] {S2};
        \draw (one) to[short, *-o] ++(0,1);
\end{circuitikz}
\end{document}
```

[![output][1]][1]

For the rotation of labels, there is almost nothing we can do --- you have  to rotate them back with `rotatebox` or `scalebox` (see section 3.2.1 in the manual, and 3.27.2).

  [1]: https://i.stack.imgur.com/c8FNy.png

Rmano, 2019-07-03:
I am closing it, please reopen if needed.

closed 2019-07-03 by Rmano