circuitikz/circuitikz #248: Volage direction arrow not inverted with 'invert'

agisen, 2019-07-04:
When inverting an european voltage source, the voltage arrow is not inverted as well (as I would have expected):
```latex
\begin{circuitikz}[european voltages, voltage dir=EF]
	\draw (0,0) to [american voltage source, v=~] (2,0);
\end{circuitikz}
\begin{circuitikz}[european voltages, voltage dir=EF]
	\draw (0,0) to [american voltage source, v=~, invert] (2,0);
\end{circuitikz}
```
(using american voltage sources to clarify the inverting)

I know that I can change the arrow direction using <, but is this behaviour intended?

Rmano, 2019-07-04:
Yes, it's intended. Or at least expected: 

[image]

(Pag. 95 in the manual).

agisen, 2019-07-04:
Ok, thanks!

closed 2019-07-04 by agisen