circuitikz/circuitikz #253: Drawing Multiplexer in Circuitikz

mickey4u, 2019-07-14:
I just want to find out how do you draw multiplexer using circuitikz.

Rmano, 2019-07-14:
It depends on what symbol you want for a multiplexer. I would use a chip as explained in the manual for the generic block, or a multi way rotative switch. What have you tried?

mickey4u, 2019-07-14:
@Rmano I saw post where someone said using the circuits.logic.IEC library offers an and gate that gives you as many input anchors as required. I just want something like the image below.
[image]

Rmano, 2019-07-14:
There is no such a symbol in circuitikz, but I think it's quite easy to do by hand, using the trapezoid shape of tikz. By the way, multiple and ports are available in circuitikz too (look at the manual of new versions), but with a quite different shape.

mickey4u, 2019-07-14:
@Rmano I will just use this one for now [https://tex.stackexchange.com/a/180351](url).

Rmano, 2019-07-15:
Closing, please reopen if you need it.

closed 2019-07-15 by Rmano