circuitikz/circuitikz #256: buggy line on rotated node!

mohammadsdtmnd, 2019-07-26:
The below mentioned code will cause bug between element m3 and a4, using miktek+texstudio.
how ever i could overcome this by using not anchor of element as reference fo path drawing and manually set coordinate.

\begin{circuitikz}[>=Stealth]  
	\draw (0,0) node[rxantenna,xscale=-1] (P1) {};
	\draw (P1.1) node[ampshape,anchor=west] (a1) {};
	\draw [->] (P1.1) ++(-1,0) -- (a1.west);
	\draw (a1.east) ++(1,0) node[mixer,anchor=west] (m1) {};
	\draw [->] (a1.east) -- (m1.west);
	\draw (m1.east) ++(2,0) node[ampshape,anchor=west] (a2) {};
	\draw [->] (m1.east) -- (a2.west);
	\draw (a2.east) ++(1,0) node[mixer,anchor=west] (m2) {};
	\draw [->] (a2.east) -- (m2.west);%اتصال میکسر به تقویت‌کننده
	\draw (m2.east) ++(1,0) node[adcshape,anchor=west] (adc) {};%نصب کانورتر
	\draw [->] (m2.east) -- (adc.west);%اتصال میکسر به کانورتر
	
	\draw (m1.south) ++(1,-1.5) node[oscillator,anchor=west] (lo) {};%نصب نوسانساز
	\draw [->] (lo.west) -| (m1.south);%اتصال میکسر به نوسانساز
	
	\draw (lo.west) ++(-1,-1.5) node[mixer,anchor=north] (m3) {};%نصب mixer
	\draw [->] (lo.west) -| (m3.north);%اتصال میکسر به نوسانساز
	\draw (m3.west) ++(-1,0) node[ampshape,anchor=west,rotate=180] (a3) {};%نصب تقویت‌کننده
	\draw [->] (m3.west) -- (a3.west);%اتصال میکسر به تقویت‌کننده
	\draw (a3.east) node[txantenna,xscale=-1] (P2) {};%نصب آنتن
	
	\draw (m3.east) ++(2,0) node[ampshape,xscale=-1,anchor=east] (a4) {};%نصب تقویت‌کننده4
	\draw [->] (a4.2) -- (m3.east);%اتصال تقویت‌ککننده به میکسر

\end{circuitikz}

Rmano, 2019-07-26:
I am not sure what the problem is, but I *think* I have seen it. The whole diagram renders like this:

[image]

and I think your problem is there: 

[image]

In this case, the solution is easy: change 

     \draw [->] (a4.2) -- (m3.east);

with 

     \draw [->] (a4.east) -- (m3.east);
    
The `amp` is a bipole, and it has **no** `2` anchor; when you say `a4.2` it means an anchor at 2 degrees from the center, starting from a rectangular ideal block. Notice that give that you use `xscale=-1` the anchors are mirrored, so 2 degrees are really 180-2=178 degrees... which is what you see. 

[image]

Do this solve your problem, or I have misunderstood you?

mohammadsdtmnd, 2019-07-26:
> \draw [->] (a4.east) -- (m3.east);

Oh, yes man this is correct, but I think that is bug since that is also valid code but malfunctioning, isn't that?

mohammadsdtmnd, 2019-07-26:
Also another weird thing happens by the last line of this code:

> \begin{circuitikz}[>=Stealth] %هواپیمایی کردن فلش‌ها در آپشن%شروع محیط 
	\draw (0,0) node[rxantenna,xscale=-1] (P1) {};%نصب آنتن
	\draw (P1.1) node[ampshape,anchor=west] (a1) {};%نصب تقویت‌کننده
	\draw [->] (P1.1) ++(-1,0) -- (a1.west);%اتصال آنتن به تقویت‌کننده
	\draw (a1.east) ++(1,0) node[mixer,anchor=west] (m1) {};%نصب میکسر
	\draw [->] (a1.east) -- (m1.west);%اتصال میکسر به تقویت‌کننده
	\draw (m1.east) ++(2,0) node[ampshape,anchor=west] (a2) {};%نصب تقویت‌کننده
	\draw [->] (m1.east) -- (a2.west);%اتصال میکسر به تقویت‌کننده
	\draw (a2.east) ++(1,0) node[mixer,anchor=west] (m2) {};%نصب میکسر
	\draw [->] (a2.east) -- (m2.west);%اتصال میکسر به تقویت‌کننده
	\draw (m2.east) ++(1,0) node[adcshape,anchor=west] (adc) {};%نصب کانورتر
	\draw [->] (m2.east) -- (adc.west);%اتصال میکسر به کانورتر
	
	\draw (m1.south) ++(1,-1.5) node[oscillator,anchor=west] (lo) {};%نصب نوسانساز
	\draw [->] (lo.west) -| (m1.south);%اتصال میکسر به نوسانساز
	
	\draw (lo.west) ++(-1,-1.5) node[mixer,anchor=north] (m3) {};%نصب mixer
	\draw [->] (lo.west) -| (m3.north);%اتصال میکسر به نوسانساز
	\draw (m3.west) ++(-1,0) node[ampshape,anchor=west,rotate=180] (a3) {};%نصب تقویت‌کننده
	\draw [->] (m3.west) -- (a3.west);%اتصال میکسر به تقویت‌کننده
	\draw (a3.east) node[txantenna,xscale=-1] (P2) {};%نصب آنتن
	
	\draw (m3.east) ++(2,0) node[ampshape,xscale=-1,anchor=east] (a4) {};%نصب تقویت‌کننده4
	\draw [->] (a4.east) -- (m3.east);%اتصال تقویت‌ککننده به میکسر
	\draw (a4.west) ++(1,0) node[oscillator,anchor=west](vco){};%نصب نوسانساز
	\draw [->] (vco.west) -- +(a4.east);
	
	
	
	
	
	
	
	
	
	
\end{circuitikz}

Rmano, 2019-07-26:
> Oh, yes man this is correct, but I think that is bug since that is also valid code but malfunctioning, isn't that?

No, I don't thik; the border anchor at 2 degrees is perfectly legal (try using `. 45`) so there is no way to discerning the two cases. Look up border anchors in TikZ manual.

mohammadsdtmnd, 2019-07-26:
> Oh, yes man this is correct, but I think that is bug since that is also valid code but malfunctioning, isn't that?
> 
> No, I don't think; the border anchor at 2 degrees is perfectly legal (try using .45) so there is no way to discerning the two cases. Look up border anchors in TikZ manual.

I can't understand what you mean, but thanks for your help.

Rmano, 2019-07-26:
> I can't understand what you mean, but thanks for your help.

I was saying that the anchor `a4.2` is working correctly. It identify a point situated on the ideal rectangle surrounding the element which is intersected by a line exiting from the center of the element at an angle of 2 degrees. I understand that in this case is not so useful, but it is working as designed.

If you want to see what I mean, try changing the 2 with, for example, 30 or 120.

mohammadsdtmnd, 2019-07-26:
> I can't understand what you mean, but thanks for your help.
>
> I was saying that the anchor a4.2 is working correctly. It identify a point situated on the ideal rectangle surrounding the element which [...]

God bless you man, you're all right.

closed 2019-07-26 by mohammadsdtmnd