circuitikz/circuitikz #258: Better CI / reference images

asinghvi17, 2019-07-31:
I'm an active contributor to JuliaPlots/Makie.jl, and I just improved the test infrastructure.  It seems quite plausible that we could apply that same framework to CircuiTikz, and create two things:
- An example gallery, like that in http://juliaplots.org/MakieReferenceImages/gallery, but for Circuitikz code - perhaps some examples from the manual could be put there, as well as drawings of each of the components, and scaling/transforming them.  This is Web-based, and can be hosted on Github Pages and accessed through an iframe.
- A reference image repository, which will be versioned separately from the main repo.  This will contain the example images, and can be used to perform visual regression tests on CI.  These tests will inform us if anything goes wrong, or what results a particular change touches.

For this, I'd probably use GItlab CI, since it offers better build artifact support - Gitlab hosts them rather than you having to configure some other hosting service.

What do you think?  @Rmano @sistlind @der-stefan @mredaelli

Rmano, 2019-07-31:
It could be a nice idea --- if the maintenance is not a burden!. 

The visual comparison would be nice, yes --- but complete coverage is quite difficult.

asinghvi17, 2019-07-31:
You could automate it - just feed in a list of every component, and autogenerate anchor descriptions (if possible), etc. 

Maintainance could even be automated on Gitlab by manually triggering a build stage.

Rmano, 2021-01-06:
Closing this issue due to inactivity; please feel free to reopen if needed.

closed 2021-01-06 by Rmano