circuitikz/circuitikz #268: Release a 0.9.4 with styling

Rmano, 2019-08-26:
Steps for 0.9.4
- [ ] freeze the repo (just a bit) 

When ready, prepare a PR in my fork  that: 

- [ ] set version number and date in circtuitiks.sty, t-circuitikz.sty, manual and changelog
- [ ] commit everything and publish the PR

Once merged, in the main repo:

- [ ] run "make ctan" this will produce a zipfile. check the contents of this file (especially the version & date of docs & package)
- [ ] go to page https://ctan.org/pkg/circuitikz
- [ ] click on upload(at the top)
- [ ] upload the zipfile and fill the missing fields ( I copied the changelog to the announcement field)

Now the repository is freezed; I'll wait for:

- [ ] Confirmation of the correct upload 
- [ ] If the package is ok it will take a few days to spawn over all servers

And after that:

- [ ] Push your commits in the main repor to github
- [ ] Add a github release with same infos & zip file.
- [ ] Add at the the link to the circuitikz-git file of the new release to the circuitikz github pages site (edit https://github.com/circuitikz/circuitikz/blob/gh-pages/index.html manually; remember to add the styles `.zip` if creating a checkpoint for this release)

[image]

- [ ] reinsert the git headline in the changelog file.

...and now development will continue.

Rmano, 2019-08-28:
Hmmm... changes are quite big, and styling is a clearly user-visible one... @mredaelli @asinghvi17 @jboalml @jesseopdenbrouw @sistlind @der-stefan : should we jump to `1.0.0`?

jesseopdenbrouw, 2019-08-28:
I agree with @asinghvi17 -- first bug hunting, that bump to 1.0.0.

Rmano, 2019-08-28:
Ok --- so that's it. Thanks for your opinion!

PR #277 (2019-09-01): Prepare for restarting development and bug fixes
Fixes #268
closed 2019-09-01 by Rmano