circuitikz/circuitikz #279: ocirc fill color

matthuszagh, 2019-09-05:
I use a dark background and am trying to fill the `ocirc` shape to that background. However, I can't seem to get it away from white. For instance

```latex
\usetikzlibrary{backgrounds}
\definecolor{fgcolor}{rgb}{0.764705882353, 0.764705882353, 0.690196078431}
\definecolor{bgcolor}{rgb}{0.133333333333, 0.133333333333, 0.133333333333}

\begin{circuitikz}
  [color=fgcolor, background rectangle/.style={fill=bgcolor},
   show background rectangle]
  \pgfsetfillcolor{bgcolor}
  \ctikzset{bipoles/length=0.8cm}
  \node at (0,0) [ground] {} coordinate(GND);
  \draw (GND) to[short, *-o] ++(-2,0) coordinate(VINm);
  \draw (GND) to[R=$R_2$] ++(0,2) coordinate(mid) to[short,*-o] ++(2,0) coordinate(VOUTp);
  \draw (mid) to[R=$R_1$] ++(0,2) to[short,-o] ++(-2,0) coordinate(VINp);
  \draw[->,>=stealth] ($ (VINm) + (0,0.25) $) -- node[right] {$V_{in}$}
    ($ (VINp) + (0,-0.25) $);
  \draw (GND) to[short, *-o] ++(2,0) coordinate(VOUTm);
  \draw[->,>=stealth] ($ (VOUTm) + (0,0.25) $) -- node[right] {$V_{out}$}
    ($ (VOUTp) + (0,-0.25) $);
\end{circuitikz}
```
gives

[image]

I used `\pgfsetfillcolor` after looking at [pgfcircshapes.tex](https://github.com/circuitikz/circuitikz/blob/f9472b09f8110c94ddc1360dfd305ba4c9d17f2d/tex/pgfcircshapes.tex#L105). I know almost nothing about pgf, so forgive me if I've made an obvious error here. How should I set this?

Is there a way to do this through `\ctikzset`? I tried a few things, but couldn't get it to work.

Rmano, 2019-09-05:
Yes --- standard `ocirc` has the white color built-in. If you look at the section 4.6 of the manual, you can find a way: redefine the poles shortcut, for example in your case (this need at least 0.9.3, better 0.9.4): 

```latex
\ctikzset{-o/.style = {bipole nodes={none}{ocirc, fill=bgcolor}}}
\ctikzset{*-o/.style = {bipole nodes={circ}{ocirc, fill=bgcolor}}}
```
will give you: 

[image]

...and as you can see, my `[ground]` has three line (ref. to issue #278).

matthuszagh, 2019-09-05:
Ok, this does indeed work. The issue is that I was using v0.8.3. Using v0.9.4 solves it. Thanks a lot for the pointers. This resolves #278 as well.

closed 2019-09-05 by matthuszagh