circuitikz/circuitikz #280: Equals sign in label causes errors

ChrisDryden, 2019-09-06:
When writing a label using the notation ```to[resistor, l={insert code here}]```

When attempting to write an equals sign inside the label, it is required to write the equals sign inside a bracket e.g. {=}. Otherwise its hard to deduce what the error is in the label and the diagram spews out random characters.

Is this something that could be fixed? I noticed its usage in the documentation.

Rmano, 2019-09-06:
It's explained in section 4.1, top of page 95 in current manual... It's a matter of how tikz processes the option list. 

[image]

Rmano, 2019-09-07:
It's a duplicate of an issue filed by me ;-) #143 --- you can find a lot of references there. Will close, re-open if you find a way to fix it!

closed 2019-09-07 by Rmano