circuitikz/circuitikz #281: Feature request: center tap transformer

asinghvi17, 2019-09-24:
It would be nice to have a center tap transformer (with an arrow pointing to the center, and a dedicated node).

For now, this can be approximated by drawing a path ending in an arrow close enough to the right inductor's midpoint.

Rmano, 2019-09-24:
Shouldn't be too difficult; the only problem I see is to precisely touch the coil, especially if the number of loops is not standard...

Rmano, 2019-09-24:
I am wondering: would an access to the internal "L" shapes suffice? Then you can really do the contact wherever you wont, like 

     ($(traf-l1.north west)!0.3!(traf-l1.north east)$)

?

Rmano, 2019-09-24:
Like: 

[image]

asinghvi17, 2019-09-24:
That would be perfect!

Rmano, 2019-09-24:
> That would be perfect!

Ok, so it's almost here. As soon as I find the time, I'll publish a PR!

PR #282 (2019-09-25): Added a "midtap" anchor to inductors; acces to inductors in trafos.
Thanks to Anshul Singhvi for suggesting it.
https://github.com/circuitikz/circuitikz/issues/281

Fixes #281 

[image]

[image]

[image]
closed 2019-09-25 by Rmano