circuitikz/circuitikz #284: Copy-paste example not compilable

suuuehgi, 2019-09-29:
I can not get a copy-paste example from the manual to compile (1st on p. 91).

```latex
\documentclass{scrartcl}

\usepackage{circuitikz}

\begin{document}

\begin{circuitikz}
    \ctikzset{multipoles/thickness=4}
    \ctikzset{multipoles/external pins thickness=2}
    \draw (0,0) node[dipchip,
        num pins=12,
        hide numbers,
        external pins width=0.3,
        external pad fraction=4 ](C){IC1};
    \draw (C.pin 1) -- ++(-0.5,0) to[R]
    ++(0,-3) node[ground]{};
    \node [right, font=\tiny]
    at (C.bpin 1) {RST};
\end{circuitikz}

\end{document}
```

Any ideas, what might be wrong?

> ! Package pgfkeys Error: I do not know the key '/tikz/circuitikz/multipoles/thi
> ckness', to which you passed '4', and I am going to ignore it. Perhaps you miss
> pelled it.
> 
> See the pgfkeys package documentation for explanation.
> Type  H <return>  for immediate help.
>  ...                                              
>                                                   
> l.8     \ctikzset{multipoles/thickness=4}
>                                          
> ? 
> 
> ! Package pgfkeys Error: I do not know the key '/tikz/circuitikz/multipoles/ext
> ernal pins thickness', to which you passed '2', and I am going to ignore it. Pe
> rhaps you misspelled it.
> 
> See the pgfkeys package documentation for explanation.
> Type  H <return>  for immediate help.
>  ...                                              
>                                                   
> l.9 ...kzset{multipoles/external pins thickness=2}
>                                                   
> ? 
> 
> ! Package pgfkeys Error: I do not know the key '/tikz/dipchip' and I am going t
> o ignore it. Perhaps you misspelled it.
> 
> See the pgfkeys package documentation for explanation.
> Type  H <return>  for immediate help.
>  ...                                              
>                                                   
> l.14         external pad fraction=4 ]
>                                       (C){IC1};
> ? 
> 
> ! Package pgfkeys Error: I do not know the key '/tikz/num pins', to which you p
> assed '12', and I am going to ignore it. Perhaps you misspelled it.
> 
> See the pgfkeys package documentation for explanation.
> Type  H <return>  for immediate help.
>  ...                                              
>                                                   
> l.14         external pad fraction=4 ]
>                                       (C){IC1};
> ? 
> 
> ! Package pgfkeys Error: I do not know the key '/tikz/hide numbers' and I am go
> ing to ignore it. Perhaps you misspelled it.
> 
> See the pgfkeys package documentation for explanation.
> Type  H <return>  for immediate help.
>  ...                                              
>                                                   
> l.14         external pad fraction=4 ]
>                                       (C){IC1};
> ? 
> 
> ! Package pgfkeys Error: I do not know the key '/tikz/external pins width', to 
> which you passed '0.3', and I am going to ignore it. Perhaps you misspelled it.
> 
> 
> See the pgfkeys package documentation for explanation.
> Type  H <return>  for immediate help.
>  ...                                              
>                                                   
> l.14         external pad fraction=4 ]
>                                       (C){IC1};
> ? 
> 
> ! Package pgfkeys Error: I do not know the key '/tikz/external pad fraction', t
> o which you passed '4', and I am going to ignore it. Perhaps you misspelled it.
> 
> 
> See the pgfkeys package documentation for explanation.
> Type  H <return>  for immediate help.
>  ...                                              
>                                                   
> l.14         external pad fraction=4 ]
>                                       (C){IC1};
> ? 
> 
> ! Package PGF Math Error: Unknown function `pin' (in 'pin 1').
> 
> See the PGF Math package documentation for explanation.
> Type  H <return>  for immediate help.
>  ...                                              
>                                                   
> l.15     \draw (C.pin 1)
>                          -- ++(-0.5,0) to[R]
> ? 
> 
> ! Package PGF Math Error: Unknown function `bpin' (in 'bpin 1').
> 
> See the PGF Math package documentation for explanation.
> Type  H <return>  for immediate help.
>  ...                                              
>                                                   
> l.18     at (C.bpin 1)
>                        {RST};

suuuehgi, 2019-09-29:
Issue found: The current [texlive distribution](https://www.archlinux.org/packages/extra/any/texlive-latexextra/) on Arch packages the old version `0.8.3`.
Using `\usepackage{circuitikz-0.9.3}` manually solves the issue.

suuuehgi, 2019-09-29:
Reopen because my previous comment is rather a workaround than a real solution.

Rmano, 2019-09-30:
Sorry, I do not understand where the problem is.

If you use an older `circuitikz` --- the newer examples do not work (as expected).  

Arch packages an older version (as Ubuntu does, and many other distributions do) --- also expected, that is why you can download a newer version here, install it locally (see for example https://tex.stackexchange.com/questions/1137/where-do-i-place-my-own-sty-or-cls-files-to-make-them-available-to-all-my-te ). You can even use the `circuitikzgit.sty` method to use still-unreleased versions, as explained in https://circuitikz.github.io/circuitikz/. 

What do you think `circuitikz` should do to solve the issue? Thanks for claryfying.

Rmano, 2019-10-01:
What I think is that your installation is at fault here: when using plain `circuitikz`, it loads the old one and not the new one --- and then it works with `circuitikz-0.9.3` because that file did not exist in 0.8.3 version. The new packages must stay in paths that are read *before*  the standard ones. 

Ok, I will close this. Please reopen if you think I misunderstood you.

closed 2019-10-01 by Rmano

suuuehgi, 2019-10-02:
> [...] install it locally (see for example https://tex.stackexchange.com/questions/1137/where-do-i-place-my-own-sty-or-cls-files-to-make-them-available-to-all-my-te ).

Thank you for the link! I did not think about this possibility.

> Arch packages an older version [...] --- also expected

Why is this expected in Arch within the 2019 release? To my (limited) understanding, this is clearly outdated, especially for Arch.

> What do you think `circuitikz` should do to solve the issue? Thanks for claryfying.

I assume(d) the version in the aforementioned distribution to be outdated but due to limited insight in the creation process of the texlive distribution I don't know who effectively is responsible for correcting this. Reading _"The stable version [...] is included within the major latex distributions"_ between the lines, I assumed this place to be the place to go. Thanks why I reopened the issue.

Rmano, 2019-10-02:
>  I don't know who effectively is responsible for correcting this

Basically, the distributions themselves. TeXLive releases yearly, but distribution normally "freeze" before the release, so that there is a mismatch. For example, TeXLive freezes around April, I think; the updates that come after that are on CTAN and can be updated via `tlmgr` but they will land in the next year release only. 
And Ubuntu freezes a lot before, especially for long term releases, so my 18.04 comes with TeXLive 2017...