circuitikz/circuitikz #285: Volt and ampere meter do not work in ConTeXt

TheTeXnician, 2019-09-30:
As reported on the ConTeXt mailing list https://mailman.ntg.nl/pipermail/ntg-context/2019/095886.html `ammeter` and `voltmeter` do not work in ConTeXt.

The causes are

https://github.com/circuitikz/circuitikz/blob/8db98802ef2d0fb4fc8a521598ffdbd2af9788d9/tex/pgfcircbipoles.tex#L2927

and

https://github.com/circuitikz/circuitikz/blob/8db98802ef2d0fb4fc8a521598ffdbd2af9788d9/tex/pgfcircbipoles.tex#L2949

Please use appropriate text formatting commands for the different formats.

Rmano, 2019-10-01:
Thanks for noticing. I know nothing of ConTeXt; can you suggest how to fix it?

Rmano, 2019-10-01:
Moreover, to test it, I have another problem: with the newer tikz, the code: 

```
\usemodule[tikz]
\starttext
{\bf go}
%\def\textbf#1{{\bf #1\/}}
% \starttikzpicture
% \draw (0,0) to[R] ++(2,0);
%\draw (0,0)to[ammeter, t=A, i=$i$] ++(3,0);
%\draw (0,4) to[rmeter, t=V, v>=$v$] ++(3,0) ;
%\draw (0,2) to[voltmeter, l=$U$] (++3,2);
% \stoptikzpicture
\stoptext
```

(notice, no `circuitikz` there) fails with:

```
tex error       > tex error on line 577 in file /home/romano/texmf/tex/generic/pgf/frontendlayer/tikz/tikz.code.tex: ! Undefined control sequence

\pgf@setup@shading@model ...delgrayfalse \XC@sdef 
                                                  \pgf@mod@test {\XC@tgt@mod...
\pgfdeclareverticalshading ...setup@shading@model 
                                                  \pgfutil@ifnextchar [\pgf@...
l.577 \pgfdeclareverticalshading
                              [tikz@axis@top,tikz@axis@middle,tikz@axis@bott...

```

I will check more. Maybe my local tikz installation is broken?

Rmano, 2019-10-01:
Ok, seems a problem in 3.1.3 which is fixed in 3.1.4b

Rmano, 2019-10-01:
Would adding `\let\textbf\bold` in `t-circuitikz.tex` an acceptable workaround?

hmenke, 2019-10-01:
PGF knows which format is being used.  You can conditionally define `\pgf@circ@font@bold` and use it instead of `\textbf`.
```latex
\ifpgfutil@format@is@latex
    \long\def\pgf@circ@font@bold{\textbf}
\else\ifpgfutil@format@is@plain
    \long\def\pgf@circ@font@bold#1{{\bf#1}}
\else\ifpgfutil@format@is@context
    \long\def\pgf@circ@font@bold{\bold}
\fi\fi
```

Rmano, 2019-10-02:
I have the same problem with `\boldmath`, it seems. Which is the equivalent for ConTeXt?

TheTeXnician, 2019-10-02:
`\boldmath` is not necessary in ConTeXt. `\bold` does bolden math as well. See e.g.

```
\starttext
Test $x$

\bold{Test $x$}
\stoptext%
```

Concerning font sizing there is `\setupbodyfont` which can set the font size like `\setupbodyfont[6pt]` (no `\selectfont` equivalent needed).

Rmano, 2019-10-02:
Ok --- I tried a solution to encapsulate all the font changes. It's in PR #286; please feel free to comment. I am still testing it; I think I'll squash it before committing anyway.

Rmano, 2019-10-02:
I squashed the commits, now they should be much more readable.

PR #286 (2019-10-03): Make internal font selection compatible with ConTeXt
See https://github.com/circuitikz/circuitikz/issues/285
Workaround from https://mailman.ntg.nl/pipermail/ntg-context/2019/095896.html

Fixes #285
closed 2019-10-03 by Rmano