circuitikz/circuitikz #292: The V letter is missing in the Voltometer, when it is rendered vertically at least.

csevast, 2019-10-05:
Instead of V, something not recognizable is rendered. I think that the problem raised when I updated `circuitikz' to the 0.9.4 version.

Also, what if the V letter rendered always horizontally? This is a request.

Thank you.

Rmano, 2019-10-05:
Can you please add an example code with the problem and, if possible, a screen shot?

csevast, 2019-10-05:
Trying to create an example I found out that the problem raises when I
apply Greek font encoding. I attach both the TEX and the produced PDF
files. When `\usepackage[LGR]{fontenc}' is opted out the V letter is
rendered normally. I noticed the problem when the LGR of `fontenc' package
raised other problems. I use the MikTex distribution. Can you help me with
this?

Rmano, 2019-10-05:
Please, if you can post a minimal example showing the problem, I can see if I can find a solution. I need to be able to reproduce it...

csevast, 2019-10-06:
Here you are. In this example, in both horizontal and vertical direction,
the V letter is not rendered when LGR (Greek) encoding is selected by
`fontenc' package is selected.

Rmano, 2019-10-06:
I am sorry --- no attached file.  Please put your file in a "zip" file and then attach it in your message here  by drag and dropping. I do not think that files attached by email pass through...

Rmano, 2019-10-06:
As an example i concocted this. Of course under LGR encoding `V` is not a `V`, but something else. If you want a `V` you have to switch:

```latex
\documentclass{article}
\usepackage[T1, LGR]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[siunitx, RPvoltages]{circuitikz}

\begin{document}
    \begin{circuitikz}
        \fontencoding{T1}\selectfont
        \draw (0,0) to [voltmeter] ++(2,0) to[voltmeter] ++(0,2);
        \draw (3,0) to [rmeterwa, t=V] ++(2,0) to[rmeterwa, t=V] ++(0,2);
    \end{circuitikz}
\end{document}
```

[image]

csevast, 2019-10-06:
You are great, you solved my problem. Thank you very much.

But, I would like to ask one more question. Is there any possibility to
render a quadripolar Wattmeter, indicating terminal voltage and current (
https://www.electrical-symbols.com/electric-electronic-symbols/electrical-test-control-symbols.htm)
?

Thank you.

Rmano, 2019-10-06:
The only watt meter available is qpprobe, see page 48 of the current manual. If you want a round symbol with generic things in it, you can use the node for or rmeter or rmeterwa and the geographical anchors.

Rmano, 2019-10-06:
For example: 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}
    \draw (0,0) to[short, -*] (1,0) to[R] (3,0)
        to[rmeterwa, t=W, name=W, -*] (5,0) -- (6,0);
    \draw (1,0) -- ++(0,1) -| (W.north);
    \draw (5,0) -- ++(0,-1) -| (W.south);
\end{circuitikz}
\end{document}
```

[image]

Rmano, 2019-10-06:
I will close this issue now, it's not a bug in `circuitikz`. Please feel free to reopen if I misunderstood.

closed 2019-10-06 by Rmano

csevast, 2019-10-06:
I will give it a try for the Wattmeter proposal.

Thank you very much.