circuitikz/circuitikz #293: Enable segment letters for the seven segment display

mickey4u, 2019-10-06:
How do I enable segment letters for the seven segment display?

Rmano, 2019-10-06:
I am really sorry, I do not understand your question. Please:

- add a sample code you tried 
- with the result you have obtained
- and explain what you want to obtain instead.

mickey4u, 2019-10-06:
As shown in the image, I want to add the segment letters to the display.

[image]

Rmano, 2019-10-06:
So, I really do not think this is an issue; it's just a "how do I do *x*" question...  I think that a forum like https://tex.stackexchange.com/questions is more appropriate for this kind of questions; issues here are for bugs or feature requests.  

Said that, seven-segments have the `a`...`g` anchors, in the center of the corresponding segment (see manual section 3.25.1) . So you can do something like this (shown for a and b, you can fill in the rest):

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}
    \node [seven segment bits=0000000 dot empty box off] (D) {}; 
    \node [above, font=\tiny]  at (D.a) {a};
    \node [right, font=\tiny]  at (D.b) {b};
\end{circuitikz}
\end{document}
```

[image]

If I misunderstood your question, please feel free to re-open the issue.

Rmano, 2019-10-06:
If I misunderstood your question, please feel free to re-open the issue.

closed 2019-10-06 by Rmano