circuitikz/circuitikz #297: Add current transformer

alexpacini, 2019-10-14:
I noticed that the symbol for the current transformer is missing.
The symbol is the same as the traditional transformer but has only an half circle on one side to show that it is only a single turn.

I wonder if it is worth adding the symbol. I contributed in the past and it should be too complicated to make it from the transformer (just leave half a turn), but unfortunately I don't have lot of time available at the moment. Therefore I am not sure if I can make it in a reasonable time (and do it properly with docs, etc.).
If anyone is willing to do that I will appreciate and will say thanks :)

Rmano, 2019-10-14:
Could you please add an example? I am not sure to understand the description...

alexpacini, 2019-10-14:
Sure. I apologise for the low quality...
The style should be the same as the transformers, but with a single turn.

[image]

Rmano, 2019-10-15:
I am thinking to allow styling of the inductors, so that you can have diagrams like these: 

[image]

alexpacini, 2019-10-15:
That is very nice.
Consider though that the current sensing side (the one with a single loop) is often placed inline with the line, therefore without the two "knees" (similarly to my picture).

Rmano, 2019-10-15:
This could be achieved with the `quadpoles style=inline` (which I prefer anyway): 

[image]

Rmano, 2019-10-15:
Hmm... there is a bug in the core of the `inline` `transformer core` --- will check.

Rmano, 2019-10-15:
> Hmm... there is a bug in the core of the `inline` `transformer core` --- will check.

Will be fixed by #298

alexpacini, 2019-10-15:
Perfect!

Rmano, 2019-10-16:
> Do you also need to set the coil width?

Yes, because the number of coils and the width are independent. But you can easily define a style: 

[image]

closed 2019-10-17 by Rmano