circuitikz/circuitikz #306: lualatex compatibility

matthuszagh, 2019-10-30:
I'm getting unexpected output when using lualatex as an engine instead of pdflatex. In particular, it seems like `\ctikzset` is having no effect. Here's an example

```
\documentclass{standalone}
\usepackage[usenames]{color}

\usepackage{circuitikz}
% color definitions
\definecolor{bgcolor}{rgb}{0.133333333333, 0.133333333333, 0.133333333333}
\definecolor{fgcolor}{rgb}{0.764705882353, 0.764705882353, 0.690196078431}
% Circuitikz style options
\ctikzset{-o/.style = {bipole nodes={none}{ocirc, fill=bgcolor}}}
\ctikzset{o-/.style = {bipole nodes={ocirc, fill=bgcolor}{none}}}
\ctikzset{*-o/.style = {bipole nodes={circ}{ocirc, fill=bgcolor}}}
\ctikzset{o-*/.style = {bipole nodes={ocirc, fill=bgcolor}{circ}}}
\ctikzset{resistors/scale=0.6, capacitors/scale=0.6, diodes/scale=0.4}

\begin{document}
\begin{circuitikz} [color=fgcolor]
  \draw (0,0) to[short, o-] ++(1,0) coordinate(P1);
  \draw (P1) to[R=$R_1$] ++(2,0) coordinate(R1);
  \draw (R1) to[R=$R_2$] ++(2,0) coordinate(R2);
  \draw [dotted] (R2) to[short,*-*] ++(1,0) coordinate(P3);
  \draw (P3) to[R=$R_n$] ++(2,0) coordinate(Rn);
  \draw (Rn) to[short,-o] ++(1,0);
\end{circuitikz}
\end{document}
```

Output when compiled with: `lualatex -shell-escape -interaction nonstopmode tmp.tex`
[lualatex.pdf](https://github.com/circuitikz/circuitikz/files/3786554/tmp.pdf)

Output with: `pdflatex -shell-escape -interaction nonstopmode tmp.tex`
[pdflatex.pdf](https://github.com/circuitikz/circuitikz/files/3786559/tmp.pdf)

Is lualatex a supported way to use circuitikz? I'm using circuitikz version 0.9.4, so I'm a little behind the master branch. Let me know if that's an issue and I'll update.

Rmano, 2019-10-30:
Hmmm... I have an error running `lualatex` (never use `nonstopmode` while debugging!); solved by https://tex.stackexchange.com/questions/313489/error-using-pdfpages-and-lualatex.  But if I add ` \usepackage{luatex85}` to your snippet now I have: 

[image]

I am using 

    This is LuaTeX, Version 1.0.4 (TeX Live 2017/Debian)

So it seems a problem of `lualatex`...

Rmano, 2019-10-30:
PD: do not load `color`; `circuitikz` loads `tikz` which loads `xcolors`. If you need some option passed to `xcolor`, use 

     \PassOptionsToPackage{usenames}{xcolor}

before calling the packages (or even before `\documentclass`  to be on the safe side, some classes require `xcolor` too...)

Rmano, 2019-10-30:
Hmmm... Strange. I'll try later, when I'll be back to a PC.

To load a package before the document class you have to use

     RequirePackage{luatex85}

Hmmm... Puzzling. Also in TeXexchange chat people confirmed that it worked... Will definitely check.

Rmano, 2019-10-31:
I still have exactly the same result with `pdflatex` and `lualatex` here; no errors with either your code or putting the `RequirePackage{luatex85}` before everything. 

[image]

I am compiling with 
```bash
TEXINPUTS=.:../my-circuitikz/tex/: pdflatex -shell-escape issue-306-lua.tex
cp issue-306-lua.pdf pdf306.pdf
TEXINPUTS=.:../my-circuitikz/tex/: lualatex -shell-escape issue-306-lua.tex
cp issue-306-lua.pdf lua306.pdf
```

I checked with 0.9.4, 0.9.5 and the current tip, everything ok.

On the [TeX and friends chat](https://chat.stackexchange.com/transcript/message/52365402#52365402) Ulrike checked it for me and told me it works (without the `luatex85` hack) in current TeXLive. 

So if for you it's working in `pdflatex` but nor in `lualatex` I *think* you need to upgrade `lualatex`...

matthuszagh, 2019-10-31:
Your intuition was correct. Updating luatex fixes the issue. Thanks for all the help with this!

Rmano, 2019-11-01:
You're welcome, I am happy that the problem is solved.

closed 2019-11-01 by Rmano