circuitikz/circuitikz #322: flip-flop library

matthuszagh, 2019-11-20:
I'm interested in extending circuitikz with flip-flop symbols. Is there any effort on this front that I should coordinate with? Or, are there any style requirements I should be aware of besides what is mentioned in the manual?

Full disclaimer, I'm fairly busy and not sure if/when I'd be able to get to this. But I wanted to start the conversation and gauge interest. FYI, I did see the note about the feature-freeze until v1.

Rmano, 2019-11-21:
Well, the couple of flip-flop I needed to this date I made with generic `chip`s, see section 3.24.4 "chip special usage" in a recent manual. 
But yes, generic flip-flops could be nice. I would recommend 

- to make it a class apart, but with a default style (thickness, etc.) similar to logic ports 
- to try to be as generic as possible, but not too much (for example, having different sizes for different types of flip-flop like D, RS, etc. is too much, you can always use local changes). 

The most difficult part in defining new shapes is when you have a variable number of anchors (pins, etc), otherwise is quite straightforward; look for example at how electronic tubes are implemented (@jesseopdenbrouw has a quite nice and clean programming style, better than mine).

Rmano, 2019-12-12:
Ah, also, the I think that the size should match the logic ports. For example, having two NAND fitting incely if I want to send them to the JK inputs... 
I have a collegue here with some flip flop done and he will probably jump in. ;-)

Rmano, 2019-12-13:
Hmmm... I had an idea and maybe this is easier than I thought. Let's see if I can find a bit of time this week-end

Rmano, 2019-12-17:
Still trying to catch a strange thing, but almost there... 

[image]

asinghvi17, 2019-12-17:
@Rmano could you put the code up?  Would love to take a look at a formal version of a 555 (not a user macro), for example.

Rmano, 2019-12-17:
Yes --- it's still in a flushing state, but no problem. It is quite long; I'll zip it. 

[flippyfloppy.zip](https://github.com/circuitikz/circuitikz/files/3974429/flippyfloppy.zip)

Rmano, 2019-12-17:
The magical uber-expandafter thing is to test for a void key, see https://tex.stackexchange.com/questions/520794/test-if-a-pgfkey-is-empty-in-a-safe-way

Rmano, 2019-12-19:
This is my current proposal: 
[flipflops.pdf](https://github.com/circuitikz/circuitikz/files/3983322/flipflops.pdf)

Rmano, 2019-12-20:
@matthuszagh @asinghvi17  the code is up for review!

Rmano, 2019-12-21:
I've merged it, please reopen in case of problems!

PR #337 (2019-12-21): Added flip-flop library.
I know I said it was in a freeze state... can't resist to add this before 1.0.
Fixes #322 

[image]
[image]
closed 2019-12-21 by Rmano