circuitikz/circuitikz #323: feature proposal - RC Snubber for switch/transistor component

EdAlvesSilva, 2019-11-21:
Hello there,

I think it would be nice to have a RC snubber option for transistors/switches in general, in a similiar way that there is for the bodydiode option. 

This is how it looked when hardcoding it to my project, and the code I used to make it: 

With a bodydiode:
[image]

`
++(0,-1) node [npn,scale=1,bodydiode,name=npn1] {}
(npn1.E)++(0,0.31) -- ++(0.5,0) to[R]  ++(0,0.5) to[C] ($(npn1.C)+(0.5,-0.31)$) -- ++(-0.31,0)
`

Without a bodydiode:
[image]

`++(0,-1) node [npn,scale=1,name=npn1] {}
(npn1.E)++(0,0.31) -- ++(0.3,0) to[R]  ++(0,0.5) to[C] ($(npn1.C)+(0.3,-0.31)$) -- ++(-0.31,0)
`
Thanks

Rmano, 2019-11-21:
To be honest, I am not so sure about it. It could be easily done with a macro (passing the reference of a transistor node) and the new `class` scaling of components; adding it as a general option will add complexity for a not-so-common case... especially if I want to fix the standard label positioning of transistors.  

I'll see if someone else comment about this, marking "enhancement" and "discussion"  for now.

Rmano, 2019-11-28:
I'll add a couple of anchors to simplify the task. Stay tuned...

Rmano, 2019-11-29:
Like this: 

[image]

what do you think?

EdAlvesSilva, 2019-11-29:
This looks great!

PR #326 (2019-11-30): Add the anchors for bodydiodes and snubbers in transistors.
Fixes #323
closed 2019-11-30 by Rmano