circuitikz/circuitikz #331: customize ocirc fill opacity

matthuszagh, 2019-12-10:
Is it possible to customize the fill opacity of a ocirc shape? I've
tried the following but haven't had any luck:

``` tex
\ctikzset{-o/.style = {bipole nodes={none}{ocirc, fill opacity=0, text
    opacity=1}}}
```

I'd previously been using this,
``` tex
\ctikzset{-o/.style = {bipole nodes={none}{ocirc, fill=some_color}}}
```
which worked well, so maybe opacity isn't exposed in ctikzet?

I took a look through pgfcircshapes.tex and found the ocirc
declaration where the fill opacity is set, but I'm not quite sure how
to override it.

Rmano, 2019-12-10:
Basically, no; this is due to the fact that the `ocirc`  and company is drawn after the path, so if you use an opacity different from 1.0 you'll see the wire underneath, which is not ok. 

Can you please point out the problem you are trying to solve?

matthuszagh, 2019-12-10:
I use circuitikz mostly to generate isolated SVGs which I want to work
on different background colors (currently limited to dark) without
modification. Here's an example:

``` shell
$ lualatex demo.tex
$ dvisvgm --pdf demo.pdf -n -b min -c 10 -o demo.svg
```

``` tex
\RequirePackage{luatex85}
\PassOptionsToPackage{usenames}{xcolor}
\documentclass[border={0pt 1pt}]{standalone}
\usepackage{circuitikz}
%% color definitions
\definecolor{bgcolor}{rgb}{0.133333333333, 0.133333333333, 0.133333333333}
\definecolor{fgcolor}{rgb}{0.764705882353, 0.764705882353, 0.690196078431}
%% Circuitikz style options
\ctikzset{-o/.style = {bipole nodes={none}{ocirc, fill opacity=0, text
    opacity=1}}}
\ctikzset{o-/.style = {bipole nodes={ocirc, fill opacity=0, text
    opacity=1}{none}}}
\ctikzset{o-o/.style = {bipole nodes={ocirc, fill opacity=0, text
    opacity=1}{ocirc, fill opacity=0, text opacity=1}}}
\ctikzset{*-o/.style = {bipole nodes={circ}{ocirc, fill opacity=0,
      text opacity=1}}}
\ctikzset{o-*/.style = {bipole nodes={ocirc, fill opacity=0, text
      opacity=1}{circ}}}
\ctikzset{resistors/scale=0.6, capacitors/scale=0.6, diodes/scale=0.4}

\tikzset{>=latex}

\begin{document}
\definecolor{fg}{rgb}{0.760784,0.760784,0.690196}%

\nopagecolor%

{\color{fg}

\begin{circuitikz}[color=fg]
  \node at (0,0) [ground] {} coordinate(GND);
  \draw (GND) to[short, *-o] ++(-2,0) coordinate(VINm);
  \draw (GND) to[R=$R_2$] ++(0,2) coordinate(mid) to[short,*-o] ++(2,0) coordinate(VOUTp);
  \draw (mid) to[R=$R_1$] ++(0,2) to[short,-o] ++(-2,0) coordinate(VINp);
  \draw[->] ($ (VINm) + (0,0.25) $) -- node[right] {$V_{in}$}
    ($ (VINp) + (0,-0.25) $);
  \draw (GND) to[short, *-o] ++(2,0) coordinate(VOUTm);
  \draw[->] ($ (VOUTm) + (0,0.25) $) -- node[right] {$V_{out}$}
    ($ (VOUTp) + (0,-0.25) $);
\end{circuitikz}%

}

\end{document}
```

This produces a light-gray foreground and colorless background, which
works as long as the background is dark enough. I can set
`fill=bgcolor` in `ctikzset`, which allows it to work with one
background but is incorrect for other backgrounds.

Rmano, 2019-12-11:
Ok, the thing is not easy to do because of the fill-out thing. I am not really sure what you need, but I prepared a patch (and a blurb explaining the problem that will arise); would something like the following help your case? 

[image]

If you want to test it, you can just add this code to the top of your file before `\begin{document}`  (only for `ocirc` for now). 

```
\makeatletter

\ctikzset{poles/fillopacity/.initial=1.0}% better not touch it
\tikzset{poles opacity/.code={%
        \ctikzset{poles/fillopacity=#1}%
}}

\pgfdeclareshape{ocirc}{
    \anchor{center}{
        \pgfpointorigin
    }
    \savedanchor\northwest{%
        \pgf@y=\ctikzvalof{nodes width}\pgf@circ@Rlen
        \pgf@x=-\pgf@y
    }
    \anchor{center}{ \pgf@y=0pt \pgf@x=0pt }
    \anchor{east}{ \northwest \pgf@y=0pt \pgf@x=-\pgf@x  }
    \anchor{e}{ \northwest \pgf@y=0pt \pgf@x=-\pgf@x  }
    \anchor{west}{ \northwest \pgf@y=0pt }
    \anchor{w}{ \northwest \pgf@y=0pt }
    \anchor{south}{ \northwest \pgf@x=0pt \pgf@y=-\pgf@y }
    \anchor{s}{ \northwest \pgf@x=0pt \pgf@y=-\pgf@y }
    \anchor{north}{ \northwest \pgf@x=0pt }
    \anchor{n}{ \northwest \pgf@x=0pt }
    \anchor{south west}{ \northwest \pgf@y=-\pgf@y }
    \anchor{north east}{ \northwest \pgf@x=-\pgf@x }
    \anchor{north west}{ \northwest }
    \anchor{south east}{ \northwest \pgf@x=-\pgf@x \pgf@y=-\pgf@y }
    \anchorborder{
        \pgf@circ@res@left=\pgf@x
        \pgf@circ@res@up=\pgf@y
        \pgfpointborderellipse{\pgfpoint{\pgf@circ@res@left}{\pgf@circ@res@up}
        }{\pgfpoint{\ctikzvalof{nodes width}*\pgf@circ@Rlen}{\ctikzvalof{nodes width}*\pgf@circ@Rlen}}
    }

    \behindforegroundpath{

        \pgfscope
            \pgfpathcircle{\pgfpointorigin}{\ctikzvalof{nodes width}*\pgf@circ@Rlen}
            \pgfsetcolor{\ctikzvalof{color}}
            \ifx\tikz@fillcolor\pgfutil@empty
                % set the default fill color to white
                \pgfsetfillcolor{white}
                % ...but override it if the class is defined!
                % note that this element has no class, but will inherit it when used
                % into another component
                \pgf@circ@setifdefinedfill{draw, fill}{draw, fill}
            \else
                \pgfsetfillcolor{\tikz@fillcolor}
            \fi
            \pgfsetfillopacity{\ctikzvalof{poles/fillopacity}}% should be 1.0
            \pgfusepath{draw,fill}
        \endpgfscope
    }
}
\makeatother
```

Rmano, 2019-12-11:
I changed the name of the option, please see PR #332

matthuszagh, 2019-12-12:
Thanks for creating a workaround for this! Given the tradeoff I think I prefer the convenience of automatic positioning over transparency in most cases, but the option might be nice in cases where I really need it. Thanks for the added documentation as well. It explains the issue very clearly.

Rmano, 2019-12-12:
Yes, could be useful for in or out-going terminals: 

```latex
% maybe really useful only for terminals going out of the cicuit
    \draw (3.6,0) -- ++(0.2,0) node[ocirc, halfwhite, anchor=180]{};
```

[image]

...will merge.

commit 97a44d48 (2019-12-12): Merge pull request #332 from Rmano/poles-transparency

Provision for transparency in poles. Fixes #331
closed 2019-12-12 by Rmano