circuitikz/circuitikz #335: Clash with the **french** option of the **babel** package

Arkh42, 2019-12-18:
If I try to compile a simple circuitikz example with the `french` option of the `babel` package, it generates a lot of compilation errors. If I compile the same example with the `british` option instead, it works perfectly.
My first guess is that problem comes from the fact that the semicolon `;` is an active character in French, which is not the case in English. I had a similar issue when creating labels for figures (or any other thing to refer to): I could not use colons in the labels in French.
Do you have any idea to fix the problem?

Here follows the LaTeX example:
```
\documentclass{article}

\usepackage[main=british]{babel}%	Works fine
%\usepackage[main=french]{babel}%	Does not compile
%\usepackage[main=british, french]{babel}%	Does not compile

\usepackage[siunitx]{circuitikzgit}

\begin{document}
	\begin{circuitikz}[]
		\draw (0,0) to[isource] (0,3) -- (2,3)
		to[R] (2,0) -- (0,0);
	\end{circuitikz}
\end{document}
```

Rmano, 2019-12-18:
This is a known problem with `babel` --- it sets a lot of chars as active. In modern times, with Unicode, that should not be necessary anymore. To fix your problem, anyway, you need to add `\usetikzlibrary{babel}` to your preamble. 

Also, you are using the wrong encoding for French; you should add `\usepackage[T1]{fontenc}` if you want your accent to stay in place (you should have a warning about this). And while at it, the preferred encoding for documents should be Unicode-compatible, so: 

```
\documentclass{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}

% \usepackage[main=british]{babel}%	Works fine
\usepackage[main=french]{babel}%	compiles
% \usepackage[main=british, french]{babel} % compiles

\usepackage[siunitx, RPvoltages]{circuitikzgit}
\usetikzlibrary{babel}

\begin{document}
	\begin{circuitikz}[]
		\draw (0,0) to[isource] (0,3) -- (2,3)
		to[R] (2,0) -- (0,0);
	\end{circuitikz}
\end{document}
```

By the way --- sometimes the `babel` Ti*k*Z library is not enough; I normally use `\deactivatequoting` too --- just after calling babel. I use Unicode for the accented letter, never the active chars combinations, and in some languages (for example Spanish) the overloading of `<` and `>` can break things even if the `\usetikzlibrary{babel}` is used. 

See also https://tex.stackexchange.com/questions/166772/problem-with-babel-and-tikz-using-draw  and related links.

Arkh42, 2019-12-19:
Well I forgot to include the encoding in the example but I had the same issue. My apologies.

However, thank you very much for the `\usetikzlibrary{babel}`. Also thank you for the link to Stack Exchange :)

May I ask you to write that issue/solution in the future manual of `circuitikz`? I must admit that I never open the Tikz manual when I use circuitikz because I find everything I need in yours.

Rmano, 2019-12-19:
Yes, I will add something. Marked as documentation bug so that I'll not forget.

Arkh42, 2019-12-19:
Thank you very much.
By the way, I used to like circuitikz, now I've been loving it since the new release.
Great job by everyone! :)

closed 2019-12-19 by Rmano

Rmano, 2019-12-20:
> Great job by everyone! :)

Thanks!

PR #336 (2019-12-19): Notice the Babel-Tikz problems in the manual.
Fixes #335