circuitikz/circuitikz #340: New shapes for american OR ports

Rmano, 2019-12-24:
Hi all, (especially @asinghvi17 @mredaelli @der-stefan @sistlind @matthuszagh  --- several people told me that the shape of american-or logic gates and similar is strange, because they are supposed to be pointy. So I have modified the shapes, see the diagram below, with a new choice flag. 

Now, I happen to like more the new shapes --- and now I have a doubt. Should I make them the default (could be a nice visual marler for version 1.0) or not? I can always put them in my style file, granted, but... 

Please comment!

[image]

matthuszagh, 2019-12-25:
My vote would be to make the new (pointy) ones default since that’s what I’m more familiar with, but it’s not a strong opinion. Thanks for all the hard work you’ve put into this!

closed 2019-12-25 by Rmano

JoseDanielMunozFrias, 2019-12-27:
I also vote to make the pointy gates the default.

Rmano, 2019-12-29:
I have merged a version with the pointy gates as default.

jason-s, 2020-03-31:
>several people told me that the shape of american-or logic gates and similar are strange because they are supposed to be pointy.

YUCK! Where did you get this information? This is not the case. If you look at any of the major logic gate manufacturers you will see standard symbols which circuitikz is now moving away from rather that toward.

Rmano, 2020-03-31:
Well. You can go back to the old ones which just one switch in your style. I prefer the new ones because they were the ones I was used to when studying. So nothing bad happened; just more options.

Rmano, 2020-03-31:
Hmmm. I see them quite similar (maybe the ones in `circuitikz` are shorter - they mostly miss the straight part.

jason-s, 2020-03-31:
I'm filing a new issue...

PR #341 (2019-12-25): Better shapes for or, nor, xor and xnor gates.
Fixes #340