circuitikz/circuitikz #346: Labels with comma won't work

cjorssen, 2020-01-17:
I think it would be safer to code `key = {#1}` instead of `key = #1` in case `#1` has comma(s).

For instance:

```
\documentclass{standalone}
\usepackage{circuitikz}
\makeatletter
\compattikzset{inductor/.style = {\circuitikzbasekey, /tikz/to path=\pgf@circ@inductor@path, l={#1}}}
\compattikzset{L/.style = {\comnpatname inductor = {#1}}}
\ctikzset{l/.code = {
        \pgfkeys{/tikz/circuitikz/bipole/label/name={#1}}
        \ctikzsetvalof{bipole/label/unit}{}
        \ifpgf@circ@siunitx
            \pgf@circ@handleSI{#1}
            \ifpgf@circ@siunitx@res
                \edef\pgf@temp{\pgf@circ@handleSI@val}
                \pgfkeyslet{/tikz/circuitikz/bipole/label/name}{\pgf@temp}
                \edef\pgf@temp{\pgf@circ@handleSI@unit}
                \pgfkeyslet{/tikz/circuitikz/bipole/label/unit}{\pgf@temp}
            \else
        \fi
        \else
    \fi
}}
\makeatother
\begin{document}
\begin{circuitikz}
  \draw (0,0) to[L={$L,r$}] ++(2,0);
\end{circuitikz}
\end{document}
```

Rmano, 2020-01-17:
Yes, you are right. But please have a look at the manual, section 4.1 (labels and annotation), around page 110 (starting with "caveat:"). It's in part a TikZ things, in part a `circuitikz` misfeature... but it's documented!

The problem is that braces should be added not only there, but all in all assignations in `pgfcircpath.tex`, pgfcircvoltages.tex`, `pgfcirccurrent.tex`, and `pgfcircflow.tex` (at least I think). I'd like to do it, but I have really not so much time now... however I'd love to review a PR! (hint, hint... ;-) )

cjorssen, 2020-01-17:
When I have more time (late april or so), I am willing to help. As it is indeed documented, I think it can wait a bit. You can consider assigning me that task.

Rmano, 2020-01-17:
Great! You're welcome!

ArchangeGabriel, 2020-01-17:
For this specific case, `[L=${L,r}$]` works though. Of course this is not general, but it can help if someone else finds this issue.

Rmano, 2020-01-17:
Yes --- it's a bit of hit and miss. When passing the keys, TikZ strips the braces... and `circuitikz` pass the keys without re-adding them. You can use `l={{{{{{{{A=B}}}}}}}` (I think that to be on the safe side you needed 6 or 7 braces) also. It was commented somewhere on tex.stackexchange.com...

Related:   https://tex.stackexchange.com/questions/39016/label-name-in-circuitikz  https://tex.stackexchange.com/questions/61471/how-can-i-put-in-a-tikz-node-label-specified-as-an-option (this one was about TikZ)

Rmano, 2021-01-01:
I am trying a code cleanup that should make this thing easier (If I manage it). So do not work on this, ok? Not yet...

Rmano, 2021-01-05:
@cjorssen  @ArchangeGabriel  I *think* that PR #484 should solve the problem. It seems not to break anything, but well, I'd like some testing...

closed 2021-01-07 by Rmano