circuitikz/circuitikz #352: Reed switch

fiveop, 2020-01-22:
I would be grateful if someone added a symbol for a reed switch (https://en.wikipedia.org/wiki/Reed_switch) to the library.

Rmano, 2020-01-22:
[image]

Is there some better source for the symbols? Seems asymmetric here, should it be so?

Rmano, 2020-01-22:
Meanwhile, you can simulate it with 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usetikzlibrary{fit, shapes.misc}
\begin{document}
\begin{circuitikz}[
    ]
    \draw (0,0) to[nos, name=mynos] ++(3,0);
    \node [rounded rectangle, draw, fit=(mynos), inner xsep=10pt] at (mynos.center) {};
\end{circuitikz}
\end{document}
```

[image]

fiveop, 2020-01-23:
I am a layman when it comes to electronics. I just found the symbol on Wikipedia and think it makes my circuit drawing more precise. A google image search shows many variations, some symmetric some not.

Thank you for the example on how to do it by hand. With a little fiddling with the inner sep values and rotation, I got it to work in my drawing.

Rmano, 2020-01-23:
Happy to hear it helped. I have no time to code a proper component now, but I'll leave this issue open because this could be a very nice first-time contribution to the project!

Rmano, 2020-02-02:
Couldn't release 1.0 with an outstanding request... See PR #359

PR #359 (2020-02-02): Added Rheed switch
[image]

Fixes #352
closed 2020-02-02 by Rmano