circuitikz/circuitikz #356: fd amp output line width customization

matthuszagh, 2020-01-31:
Is it possible to customize the `fd op amp` node so that the output pin lines are a normal line thickness? They come thick by default. I didn't see anything about this in the manual. I guess I could achieve the same effect with a mux, but that feels unnecessary since that's the only thing I want to change from the default.

Rmano, 2020-01-31:
It shouldn't do that. I'll check as soon as I can...

Rmano, 2020-01-31:
So yes, the problem is quite evident: 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}[
    circuitikz/amplifiers/fill=yellow]
    \draw (0,0) node [fd op amp](A){} (A.out +) -- ++(1,0);
\end{circuitikz}
\end{document}
```

[image]

Rmano, 2020-01-31:
After the patch:

[image]

Notice that some PDF viewer (notably, `okular`) can get the antialiasing wrong and show the amplifiers terminal "blurred" at some magnification.... 

`okular` at 400%: 

[image]

`okular` at 800% (same file): 

[image]

closed 2020-01-31 by Rmano

matthuszagh, 2020-01-31:
thanks for the quick fix! looks great

PR #357 (2020-01-31): Make the "fd op amp" output leads respect the wire thickness.
This fixes issue #356; thanks to @matthuszagh for noticing and
reporting.

Fixes #356