circuitikz/circuitikz #361: Releasing v1.0

Rmano, 2020-02-04:
And finally... version 1.0 (2020-02-04) of `circuitikz` is released.

The main updates since version 0.8.3, which was the last release before Romano started co-maintaining the project, are the following --- part coded by Romano, part by several collaborators around the internet:

    - The manual has been reorganized and extended, with the addition of a tutorial part; tens of examples have been added all over the map.
    - Around 74 new shapes where added. Notably, now there are chips, mux-demuxes, multi-terminal transistors, several types of switches, flip-flops, vacuum tubes, 7-segment displays, more amplifiers, and so on.
    - Several existing shapes have been enhanced; for example, logic gates have a variable number of inputs, transistors are more configurable, resistors can be shaped more, and more.
    - You can style your circuit, changing relative sizes, default thickness and fill color, and more details of how you like your circuit to look; the same you can do with labels (voltages, currents, names of components and so on).
    - A lot of bugs have been squashed; especially the (very complex) voltage direction conundrum has been clarified and you can choose your preferred style here too.

A detailed list of changes can be seen in the manual. 

When ready, prepare a PR in my fork  that: 

- [ ] set version number and date in circtuitiks.sty, t-circuitikz.sty, and changelog
- [ ] commit everything and publish the PR

Once merged, in the **main repository**:

- [ ] run "make ctan" this will produce a zipfile. check the contents of this file (especially the version & date of docs & package)
- [ ] go to page https://ctan.org/pkg/circuitikz
- [ ] click on upload(at the top)
- [ ] upload the zipfile and fill the missing fields ( I copied the changelog to the announcement field)
- [ ] Add a github release with same infos & zip file & styles zip file
- [ ] Add at the the link to the circuitikz-git file of the new release to the circuitikz github pages site (edit https://github.com/circuitikz/circuitikz/blob/gh-pages/index.html manually; remember to add the styles `.zip` if creating a checkpoint for this release)

[image]

Another trick: to transform the link to a stable link, press "y" in the page showing it...  see https://help.github.com/en/articles/getting-permanent-links-to-files

Now the repository is frozen; I'll wait for:

- [ ] Confirmation of the correct upload 
- [ ] If the package is ok it will take a few days to spawn over all servers

And after that:

- [ ] Prepare a branch in my repo to reinsert the git headline in the changelog file, and do a PR.
- [ ] Merge your commits in the main repo to github

...and now development will continue.

der-stefan, 2020-02-04:
Thank you very much, Romano! Your work on CircuiTikZ is totally amazing and I am glad that you took over the development and maintenance of the project. On conferences one can easily see if someone used CircuiTikZ for his schematics - and it is increasingly used...

Rmano, 2020-02-04:
I just built upon the shoulder of giants...
 
I think that the package has its share of problems (sometimes the path-building is a bit fragile, and it's really monolithic --- should be divided into modules) but it's pretty usable now. I used it in my last paper and the circuits look really nice.

Thanks a lot for the nice words!

closed 2020-02-11 by Rmano