circuitikz/circuitikz #365: Moving forward past 1.0

Rmano, 2020-02-11:
This issue is a kind of discussion board for the future progress of `circuitikz`, and to identify how to continue development. I will start it listing things that I see could make the package better; both major ones and minor ones. 

The major one will lead, sometime, to backward incompatibilities; I do not know if in case of big ones the project should be renamed to `circuitikz2` or something like that. I do not like it, but sometimes it is needed... 

Major things: 

1. `circuitikz` now is monolithic. We should probably split the 300+ components into sublibraries, like for example `tcolorbox` or  Ti*k*Z itself. That will need to shuffle a lot of things around (definitions, basic utilities, etc.)
2. We should use the [new standard LaTeX version rollback](https://www.latex-project.org/publications/2018-FMi-TUB-tb122mitt-version-rollback.pdf) for backward compatibility --- but I do not know how to do that without breaking ConTeXt.
3. By the way, finding a co-maintainer using ConTeXt would be a really nice idea.

Minor things:
 
1. removing most of the package options, given that we have styles now. That's easier, and the incompatibilities are minor.
2. cleaning up the label positioning code a bit or a lot (largely my fault).
3. Do the same for the voltage positioning things, trying to add styling to them.
4. We should try to encapsulate the part of the code that uses internal interfaces of Ti*k*Z (basically, the dynamic anchors) so that it would be easy to change them if Ti*k*Z changes.

Any other/different thoughts?

matthuszagh, 2020-06-11:
Just chiming in, I don't personally like changing the name to `circuitikz2`. I prefer just leaving it as `circuitikz`. That's really just a stylistic preference though.

Rmano, 2020-06-11:
>  I don't personally like changing the name to `circuitikz2`.

I'd rather not change the name too; but if there are changes that will change completely the behavior for old circuits, that would be a necessity. I really like the `compat` strategy pf `pgfplots`, but it's a bit too late to implement it, I fear...

septatrix, 2020-06-12:
I too prefer leaving the name as it is uncommon in the tex world to have packages with a version number. However I am also a fan of not maintaining backwards compatibility at all costs. One very simple version could be to let one declare a version with which the document was first typeset and when you introduce breaking changes the package could show a warning and propose either updating the declared version if everything fit or download the past version and put it next to the document to use that version

closed 2020-10-02 by Rmano