circuitikz/circuitikz #366: New (inherited) components request

patrikradvanyi, 2020-02-16:
Hi there!

I would like to request two new components (all of them are modifications of something).

The first one is an 'open barrier'. The barrier symbol without an overall line. I've attached the definition based on your email. It was almost perfect, I just modified a little bit.

[openbarrier.txt](https://github.com/circuitikz/circuitikz/files/4209732/openbarrier.txt)

The second one is the impedance (should be called Z) symbol. This one is similar to the European resistor symbol, but there are two more lines. If I wanted to draw in the `circuitikz` environment it would look like this:

```
\draw
(0,0) to[R,l=$Z$,n=Z,o-o] (2,0)
(Z.nw) -- (Z.se)
(Z.ne) -- (Z.sw)
;
```

Thank you very much!

Patrik

Rmano, 2020-02-16:
Hmmm... I do not understand your modification. You changed my

```
\pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{0pt}}
    \pgfpathlineto{\pgfpoint{0.35\pgf@circ@res@left}{0pt}}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right}{0pt}}
    \pgfpathlineto{\pgfpoint{0.35\pgf@circ@res@right}{0pt}}
    \pgfusepath{draw}
```

which led to 

[image]

to:

```
% this is set with normal wire linewidth
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{0pt}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@left}{0pt}}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right}{0pt}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{0pt}}
    \pgfusepath{draw}
```

which is drawing two lines of length 0 and draw the following:

[image]

... to obtain which you could just have deleted the five lines above. 

Is there any standard reference of those symbols? Especially the "crossed" impedance is quite new for me, I have always used the `generic` shape to indicate impedance --- I understand that could be some local standard, but I never saw it since my first electronic class in 1982... ;-)

patrikradvanyi, 2020-02-16:
Hi!

Sry for my syntax (I was on a bus when I made it and I'm never used PGF before). It was just too crowded for me. I think it's not a strict thing. Maybe the right symbol is between your solution and my solution. Here is an example from a book.

[image]

About the impedance. Here is a picture too.

[image]

In Hungary, we always use these symbols. I don't know if there is any standard for our country.

Rmano, 2020-02-16:
What about this? 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\makeatletter
%
% open version of the barrier symbol
% suggested by Radványi Patrik Tamás <patrikradvanyi@gmail.com>
%
\ctikzset{bipoles/openbarrier/gap/.initial=0.5}
\pgfcircdeclarebipolescaled{misc}
{}
{\ctikzvalof{bipoles/barrier/height}}
{openbarrier}
{\ctikzvalof{bipoles/barrier/height}}
{\ctikzvalof{bipoles/barrier/width}}
{
    % this is set with normal wire linewidth
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{0pt}}
    \pgfpathlineto{\pgfpoint{\ctikzvalof{bipoles/openbarrier/gap}*\pgf@circ@res@left}{0pt}}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right}{0pt}}
    \pgfpathlineto{\pgfpoint{\ctikzvalof{bipoles/openbarrier/gap}*\pgf@circ@res@right}{0pt}}
    \pgfusepath{draw}

    % do the cross part
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}

    \pgfpathmoveto{\pgfpoint{0.35*\pgf@circ@res@left}{0.35*\pgf@circ@res@up}}
    \pgfpathlineto{\pgfpoint{0.35*\pgf@circ@res@right}{0.35*\pgf@circ@res@down}}
    \pgfpathmoveto{\pgfpoint{0.35*\pgf@circ@res@left}{0.35*\pgf@circ@res@down}}
    \pgfpathlineto{\pgfpoint{0.35*\pgf@circ@res@right}{0.35*\pgf@circ@res@up}}

    \pgfusepath{draw}
}
\def\pgf@circ@openbarrier@path#1{\pgf@circ@bipole@path{openbarrier}{#1}}
\compattikzset{openbarrier/.style = {\circuitikzbasekey, /tikz/to path=\pgf@circ@openbarrier@path}}
%
% generic crossed, suggested by Radványi Patrik Tamás <patrikradvanyi@gmail.com>
% inherit "generic" properties
%
\pgfcircdeclarebipolescaled{resistors}
{}
{\ctikzvalof{bipoles/generic/height}}
{xgeneric}
{\ctikzvalof{bipoles/generic/height}}
{\ctikzvalof{bipoles/generic/width}}
{
    \pgfpathrectanglecorners{\pgfpoint{\pgf@circ@res@left}{\pgf@circ@res@up}}{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@down}}
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgf@circ@draworfill
    % cross it
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{\pgf@circ@res@up}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@down}}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@up}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@left}{\pgf@circ@res@down}}
    \pgfusepath{draw}
}
\def\pgf@circ@xgeneric@path#1{\pgf@circ@bipole@path{xgeneric}{#1}}
\compattikzset{xgeneric/.style = {\circuitikzbasekey, /tikz/to path=\pgf@circ@xgeneric@path, l=#1}}
\makeatother
\begin{document}
\begin{circuitikz}[
    ]
    \draw (0,1) to[openbarrier, bipoles/openbarrier/gap=0.9] ++(2,0);
    \draw (0,0) to[openbarrier] ++(2,0);
    \draw (0,-1) to[xgeneric] ++(2,0);
    \draw (0,-2) to[generic] ++(2,0);
\end{circuitikz}
\end{document}
```

[image]

As you can see, the gap in the `openbarrier` component can be changed with a parameter. The `xgeneric` component, on the other hand, inherit the size from `generic` (it's not difficult to change, but I am not sure if it is needed).

patrikradvanyi, 2020-02-16:
Yeah :) I love it, nice job :)

Rmano, 2020-02-16:
About the open barrier I am not so sure --- maybe the right approach would be to have just one value of the gap (the 0.5 for example) and having the size of the component reduced (most of it is lead wires now...). Not sure about it.

patrikradvanyi, 2020-02-16:
Unfortunately, I don't know the opportunities and its consequences, so I just can tell my option about the result 😶

Rmano, 2020-02-16:
Finally, I think that's better if the open barrier matches the barrier component. I will merge this into `circuitikz` when I have time... till then you can just copy the code from the `makeatletter` to `makeatother` where you need it.

patrikradvanyi, 2020-02-16:
Ok, thank you very much! 😄

PR #372 (2020-02-21): Several new components for 1.0.1
Adding components

Fixes #370 
Fixes #366 

[image]

[image]

[image]
closed 2020-02-21 by Rmano