circuitikz/circuitikz #367: Circuitikz scale battery2 error

AndreaDiPietro92, 2020-02-18:
Problem with scale in circuitikz. The following code produce 

\documentclass[border=10pt]{standalone}
\usepackage[RPvoltages]{circuitikz}

\begin{document}

\begin{circuitikz}[scale=1.2, transform shape]
	\draw (12,1) to [battery2, v_=$V_{cc}$] ++(0,2);
\end{circuitikz}

\end{document}

The problem does not arise when the scale is an int value (scale=1,2,3 ..etc) or with no scale.

[image]

Rmano, 2020-02-18:
Yes, confirmed. Puzzling.

Rmano, 2020-02-18:
Oops, wrong button ;-). Reopening.

Rmano, 2020-02-18:
I suspect some precision error in the positioning of the component in the path. That will not be easy to track down...

Rmano, 2020-02-19:
If you change the coordinate `(12,1)` to `(0,1)` all is right. So it's a difficult problem... I opened a question here: https://tex.stackexchange.com/questions/529099/path-joining-in-to-path-with-non-integer-scale-factors

AndreaDiPietro92, 2020-02-19:
it's a really strange problem.

AndreaDiPietro92, 2020-02-19:
Little workaround! Using the scope enviroment for the shift.
```latex
\documentclass[border=10pt]{standalone}
\usepackage[RPvoltages]{circuitikz}

\begin{document}
	
	\begin{circuitikz}[scale=1.2, transform shape]
	\begin{scope}[shift={(11,0)}]
		\draw (1,1) to [battery2, v_=$V_{cc}$] ++(0,2);
	\end{scope}	
	\end{circuitikz}
	
\end{document}
```
[image]

Much better then before!

Rmano, 2020-02-19:
Yes, I suppose this is a kind of loss of precision in Ti*k*Z (well, in `pgfmath` basically). All calculations are done in 32-bit fixed arithmetic...
Let's see if the magical people from the TeX core can suggest a way to amend this, but I suspect it's not `circuitikz` related...

Rmano, 2020-02-19:
Well, the incredible Schrödinger's cat at https://tex.stackexchange.com/a/529159/38080 solved the issue... 

```
\documentclass[border=10pt]{standalone}
\usepackage[RPvoltages]{circuitikz}
\usetikzlibrary{fpu}
\makeatletter
%%
%% this code has been contributed by Schrödinger's cat
%% https://tex.stackexchange.com/a/529159/38080
%%
\tikzset{use fpu reciprocal/.code={%
\def\pgfmathreciprocal@##1{%
    \begingroup
    \pgfkeys{/pgf/fpu=true,/pgf/fpu/output format=fixed}%
    \pgfmathparse{1/##1}%
    \pgfmath@smuggleone\pgfmathresult
    \endgroup
}}}%
\makeatother

\begin{document}

\begin{circuitikz}[scale=1.2, transform shape,
    use fpu reciprocal,
    ]
\draw (12,1) to [battery2, v_=$V_{cc}$] ++(0,2);
\end{circuitikz}

\end{document}
```

PR #371 (2020-02-20): Fix precision loss
Add a key ([Schrödinger's cat](https://tex.stackexchange.com/a/529159/38080) idea!) to solve a loss of precision problem in fractional scalings.

Fixes #367
closed 2020-02-20 by Rmano