circuitikz/circuitikz #370: LED

karlkappe, 2020-02-20:
The arrows on the LED und photodiode, at least in the european version, are pointing in a different direction than given by DIN EN 60617 (LED: https://commons.wikimedia.org/wiki/File:Symbol_LED_IEC60617-5_Symbol_Nr_05-03-02.svg )
Stefan Lindner tild me he prefers the now implemented ones, so maybe you could add the DIN EN sysmbols also?

Rmano, 2020-02-20:
I will add a couple of flags to flip LEDs and photodiode arrows.

Rmano, 2020-02-21:
[image]

Rmano, 2020-02-21:
[image]

PR #372 (2020-02-21): Several new components for 1.0.1
Adding components

Fixes #370 
Fixes #366 

[image]

[image]

[image]
closed 2020-02-21 by Rmano