circuitikz/circuitikz #374: Positioning node voltages correctly using the american voltage

rhandley, 2020-02-27:
I recently had difficulty in correctly positioning the open circuit voltage labels between two nodes using the american voltage notation . This was reported in [https://tex.stackexchange.com/questions/529626/how-to-position-open-node-voltages-correctly], when usign the to[open, v=...]. The only solution so far is to manually position the labels "+" and "-" to the desired position. There was also an advice to open an issue as a feature request for this.

Rmano, 2020-02-27:
The request is logical (position of american-style voltage markers on open circuits has no sense now). I will try to tackle this (I am thinking about revising the voltage label thing anyway, it's a bit of a mess now, also my fault).

Rmano, 2020-04-18:
I looked at it, but it's quite difficult, and a very big rewrite would be needed. Hmm...

PR #400 (2020-05-17): Fix position of open circuit american voltages
Also, sort out the mess with straigth voltages and add them to
the manual.

Fixes #374

[image]
closed 2020-05-17 by Rmano