circuitikz/circuitikz #375: Add Schottky transistors

jmonclard, 2020-02-29:
I would be interested in Schottky transistors symbols (a bipolar transistor with an S-shaped base similar to the schottky diode), in normal and multi-terminal variants.
That can be found in (mature) TTL-S and TTL-LS logic families.
Probably a straightforward addition to the library.
BTW : Thanks a lot for your so useful package !

Rmano, 2020-02-29:
It is not difficult --- I'll see if I can find a moment to do it.
If someone is interested, I think that the best way is to define a new `\ifpgfcircdoschotty` and then add it where the base is drawn (but I'm open to other solutions...).

Rmano, 2020-02-29:
[image]

What do you think?

jmonclard, 2020-02-29:
That's great ! Thank you so much for having been so amazingly reactive.
I will look forward for the next package update, or look at the change in your fork and patch my tex if I need to complete my document sooner.

Rmano, 2020-02-29:
Nice to know you like it. You can also use the `...git` version at https://circuitikz.github.io/circuitikz as soon as I merge the PR...

PR #377 (2020-02-29): Adding Schottky transistors
Suggested by Jérôme Monclard on GitHub

Fixes #375 

[image]
closed 2020-02-29 by Rmano