circuitikz/circuitikz #378: Inerter

htadashi, 2020-03-10:
First of all, I'd like to thanks all the contributors to this excellent package! 

I'd like to suggest adding the [inerter](https://ieeexplore.ieee.org/document/1039800)¹ as a new component to the Mechanical Analogy section. I don't know if this is already the default symbol of the inerter, but most of the papers that I've seen use the notation described in this following MWE:

```
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}

\makeatletter
% used to process styles for to-path
\def\TikzBipolePath#1#2{\pgf@circ@bipole@path{#1}{#2}}
\makeatother
\newlength{\ResUp} \newlength{\ResDown}
\newlength{\ResLeft} \newlength{\ResRight}

% inerter
\ctikzset{bipoles/inerter/height/.initial=.80}
\ctikzset{bipoles/inerter/width/.initial=.30}
\pgfcircdeclarebipole{}
{\ctikzvalof{bipoles/inerter/height}}
{inerter}
{\ctikzvalof{bipoles/inerter/height}}
{\ctikzvalof{bipoles/inerter/width}}
{
\pgfsetlinewidth{\pgfkeysvalueof{/tikz/circuitikz/bipoles/thickness}\pgfstartlinewidth}
\pgfextractx{\ResRight}{\northeast}
\pgfextractx{\ResLeft}{\southwest}
\pgfextracty{\ResDown}{\southwest}
\pgfpathrectanglecorners{\northeast}{\southwest}
\pgfusepath{draw} %draw inerter
}
\def\inerterpath#1{\TikzBipolePath{inerter}{#1}}
\tikzset{inerter/.style = {\circuitikzbasekey, /tikz/to path=\inerterpath, l=#1}}

\begin{document}
\begin{circuitikz}
    \draw (0,0) to[inerter] ++(2,0);
\end{circuitikz}
\end{document}
```

[1]: For those without access to IEEE, a presentation of this paper is available without paywall here: http://www-control.eng.cam.ac.uk/foswiki/pub/Main/MalcolmSmith/lecture_j.pdf

Rmano, 2020-03-10:
The symbol is this one: 

[image]

It can be added --- especially if you want it to follow the class properties of `mechanical`, but it can be easily done styling "generic":

```latex
\tikzset{myinerter/.style={generic,
        bipoles/generic/width=0.3,
        bipoles/generic/height=0.8,
}}

\begin{document}
\begin{circuitikz}
    \draw (0,0) to[inerter] ++(2,0) to[myinerter] ++(2,0);
\end{circuitikz}
```

[image]

htadashi, 2020-03-24:
@Rmano 

Thanks for the reply, your solution is much more simpler. Pull request (#380 ) adds the inerter to the tex files and to the manual, which compiled correctly in my system.

Best regards!

closed 2020-03-25 by Rmano

Rmano, 2020-03-25:
Merged in! It will be in 1.0.3 release. Thank you and keep contributing!