circuitikz/circuitikz #383: Improve logic gate shapes (American OR/AND/XOR) to match standard symbols

jason-s, 2020-03-31:
Please improve the OR/AND/XOR symbols in circuitikz. Right now they are not consistent with the symbols used in semiconductor manufacturer datasheets for these gates, or the standard schematic capture symbols. The circuitikz versions look more stylized, like Star Trek insignia.

Suggestions for reputable sources:
- Semiconductor manufacturers of logic gates
    - TI
    - Nexperia (formerly Philips)
    - On Semi (formerly Motorola)
- Schematic capture programs
    - Altium
    - OrCAD
    - Mentor Graphics / DxDesigner

Rmano, 2020-04-01:
Very nice research work, thanks. 

I think I can try a shoot at this when I have some time (really swamped now with the coronavirus emergency and the on-line shift of classes and everything). Obviously this must be done under a style flag because the change in proportions is quite big. 

I think that we can leave the position of terminals like now (more or less it matches) and the thickness is already configurable. 

I have a couple of doubts about implementation. 

1. now the size of "not" circles is the same as poles in the circuits --- to match the rest of them (and the circle in PMOS transistors). I am thinking of leaving them alone. 
2. in current gates the output "not" circle in included *into* the width of the gate, so that stacking vertically the gates with labels at the output has that labels naturally align. In the specs above the dot is sticking out --- having the two things at the same time (dot sticking out and label alignment) is going to be very complex. I do not know where to give here.
3. (more important) I do not know how to scale the ports for a lot of input ports (say 16, which is the maximum now). The current design has independent width and height, so you can do tricks (see the manual); this spec has just one dimension (the usage of circles and not splines clearly forces that) and so a 16-input port with a standard input pin distance will be unbearably wide (see t[he example on the manual](http://circuitikz.github.io/circuitikz/circuitikzmanualgit.pdf) around page 101).

I will mention the co-authors and recent contributors here so that they can chime in --- @mredaelli  @sistlind @asinghvi17 @der-stefan @jesseopdenbrouw

jason-s, 2020-04-01:
see https://gist.github.com/jason-s/ebea0a9d5fae9a7d7f5eae2f9620f98e

I don't know how to properly make circuitikz shapes, there were a couple of fudged (empirical) numbers I put in to make it look ok. Otherwise I have something for my purposes and I am fine with circuitikz using it in whichever way you wish.

[image]

jason-s, 2020-04-01:
>I think I can try a shoot at this when I have some time (really swamped now with the coronavirus emergency and the on-line shift of classes and everything). Obviously this must be done under a style flag because the change in proportions is quite big.

yeah, I'm in no hurry; again, I have what I need to make my own diagrams.

I'd suggest `ieee` as a style flag.

> in current gates the output "not" circle in included into the width of the gate, so that stacking vertically the gates with labels at the output has that labels naturally align. In the specs above [...]

Not sure what to recommend. (I didn't implement any of the dotted variants.) Let me see if I can find an example in corporate datasheets; the IEEE standards don't show the labels.

The XOR input curve is a related issue; it's not clear whether to extend the input terminals so they protrude a consistent length from the gate, or whether the overall width should be the same for all three. (I favor the latter, so I used it in my implementation.)

jason-s, 2020-04-01:
>I will mention the co-authors and recent contributors here so that they can chime in --- @mredaelli @sistlind @asinghvi17 @der-stefan @jesseopdenbrouw

Is there an online channel (gitter or Teams or discord or Google Groups or whatever) for tikz or circuitikz development? I'd like to lurk there so that I can see what's going on, rather than watch all the github issues.

jason-s, 2020-04-01:
OK, for examples of wide-input gates in industry, see (⚠️  under construction)

- 74xx133 13-input NAND gate
    - [TI SN74ALS133](http://www.ti.com/lit/ds/sdas202b/sdas202b.pdf) 
    - [National Semiconductor MM74HC133](https://pdf1.alldatasheet.com/datasheet-pdf/view/100973/NSC/MM74HC133.html)
- 74xx4078 8-input NOR gate
    - [ST M74HC4078](https://www.mouser.com/datasheet/2/389/stmicroelectronics_cd00000322-1204332.pdf)

SN74ALS133
[image]

MM74HC133
[image]

M74HC4078 (crowded together)
[image]

Rmano, 2020-04-01:
Wow! Nice information (and thanks for the code, it'll be useful). I'll look into it; the extension for several inputs is not a bad idea... 

Yes, `ieee logic ports` would be a nice option to have. Well. I'll see what I can do. Maybe the not dot can be added after the fact, as the input ones --- I could just use your code in this case. Maybe they could just be a different family of components that can have matched pin widths similar to what muxdemuxes are.   

> Is there an online channel (gitter or Teams or discord or Google Groups or whatever) for tikz or circuitikz development? I'd like to lurk there so that I can see what's going on, rather than watch all the github issues. I had no idea that #340 was in progress, and if I had been aware of it then, I would have spoken up.

No, there isn't; basically everything goes through this GitHub page. I lurk often on [TeX.sx chat room](https://chat.stackexchange.com/rooms/41/tex-latex-and-friends) (depending on the available time).

jason-s, 2020-04-01:
As far as the logic bubbles / shape widths go, here are some examples in the wild showing that in the majority of cases, the inversion bubbles do not change the geometry, and can be inserted/removed without changes:

[TI SN74LVC1G99](http://www.ti.com/lit/ds/symlink/sn74lvc1g99-q1.pdf)
[image]

[image]

[NXP 2016 Logic Selection Guide](https://www.digikey.com/Site/Global/Layouts/DownloadPdf.ashx?pdfUrl=755DA3330E4E422EAEF6B9FF5092A9CB)

[image]

[Nexperia 74LVC1G99](https://assets.nexperia.com/documents/data-sheet/74LVC1G99.pdf) -- the text/connections have geometry independent of the bubbles, but note the Schmitt trigger symbol which moves to be as close to the symbol outline as possible... and it looks weird to me.

[image]

jason-s, 2020-04-01:
> the extension for several inputs is not a bad idea...

that wasn't from me, that was from the tex.stackexchange.com post. I'd suggest doing a diff between my gist and the original (i'll add it to the gist) -- I didn't change much.

Rmano, 2020-04-01:
> > the extension for several inputs is not a bad idea...
> 
> that wasn't from me, that was from [John Kormylo's tex.stackexchange.com post](https://tex.stackexchange.com/a/372996/46076) which I _thought_ had been incorporated into circuitikz already. I'd suggest doing a diff between my gist and the original (i'll add it to the gist) -- I didn't change much.

Yes --- I have incorporated it (with minor changes) since 0.9.0 I think. I was referring to the vertical bar used for more than two o three inputs that you posted from the datasheets:

 
[image]

Rmano, 2020-04-08:
I will assign this to me. Trying to tackle it during Easter (...nowhere to go...)

Rmano, 2020-04-08:
@jason-s  --- any special way you prefer for citing you in the code and documentation? `Jason Sachs<any email here>`?

Rmano, 2020-04-08:
One thing. In the table the NOT port seems more or less the same height than a 2-inputs NOR:
[image]

...but it seems quite big to me. Have you any example with 2-inputs and NOT ports togheter?

jason-s, 2020-04-09:
>@jason-s --- any special way you prefer for citing you in the code and documentation? Jason Sachs<any email here>?

Jason Sachs <jmsachs@gmail.com>

jason-s, 2020-04-09:
> One thing. In the table the NOT port seems more or less the same height than a 2-inputs NOR:
> [image]
> 
> ...but it seems quite big to me. Have you any example with 2-inputs and NOT ports togheter?

Neither IEEE 91-1973 nor IEEE 91-1984 recommends dimensions for a NOT gate, which is kind of odd, but see the examples I posted above for the 1G99 gates from Nexperia and TI, where the NOT gate is the same height.

[Nexperia 1G99](https://assets.nexperia.com/documents/data-sheet/74LVC1G99.pdf):

[image]

[TI 1G99](http://www.ti.com/lit/ds/symlink/sn74lvc1g99-q1.pdf)

[image]

Rmano, 2020-04-09:
Yes, I think that the standard is the same height. Thanks!

When I have something test-able, I'll post here.

Rmano, 2020-04-11:
I am at a good point. I have 

[image]

and anchors do work...

[image]

...stay tuned!

Rmano, 2020-04-11:
@JoseDanielMunozFrias --- do you like them?

JoseDanielMunozFrias, 2020-04-13:
¡They are prefect!

¡Great work!

Rmano, 2020-04-13:
Current state of the manual:

[ieeeports-manual.pdf](https://github.com/circuitikz/circuitikz/files/4467103/ieeeports-manual.pdf)

jesseopdenbrouw, 2020-04-13:
Perfect & marvelous!

Rmano, 2020-04-13:
I think that the not-port should be equilater triangles... will check. Not too difficult to change ;-).

jason-s, 2020-04-13:
> Current state of the manual: [ieeeports-manual.pdf]
> 

Two suggested corrections:

>In addition to the legacy ports, since release 1.0.3, also logic port conforming (more or less) to
IEEE Std 91/91a-1984 and IEEE Std 91/91a-1991 are available. The suggestion and a first shot
at the code is bt Jason Sachs (jmsachs@gmail.com); you can see it on this GitHub issue

The IEEE standards don't have requirements of how these basic symbols should look, only recommendations. The requirements of these standards have more to do with annotations on digital logic, like certain designators: `BI` = borrow input, `BO` = borrow output, `CI` = carry input, `CO` = carry output, D/J/K/R/S for flip-flops and latches and what they are supposed to mean, etc. And the vast majority of the standard applies to the Rectangular-shape symbols, not the Distinctive-shape symbols. The "Annex A" (Recommended symbol proportions) just covers *recommended* geometric dimensions for the both the Rectangular- and distinctive-shape symbols.

The rectangular-shape symbols you will see in datasheets also, especially when there are more complex behaviors; here's from the [TI 74LS138](https://www.ti.com/lit/ds/symlink/sn54ls138-sp.pdf):

[image]

So I would remove the word `confirming` and instead suggest this wording:

>In addition to the legacy ports, since release 1.0.3, logic ports following the recommended geometry of distinctive-shape symbols in IEEE Std 91a-1991 Annex A (Recommended symbol proportions)  are also available.

Second issue: I appreciate the credit but please don't put my name/address there in the body of the document; in the release notes or in the contributors list is fine.

jason-s, 2020-04-13:
> I think that the not-ports should be equilateral triangles... will check. Not too difficult to change ;-).

Yeah, it looks that way to me, although the IEEE Std 91-1984 and 91-1973 don't include not gates in their Appendix/Annex A for some reason. (Amplifiers are included, and they are equilateral triangles somewhat larger than the logic gate symbols.)

Rmano, 2020-04-16:
> So I would remove the word `conforming` and instead suggest this wording:
> 
> > In addition to the legacy ports, since release 1.0.3, logic ports following the recommended geometry of distinctive-shape symbols in IEEE Std 91a-1991 Annex A (Recommended symbol proportions)  are also available.
> 
> Second issue: I appreciate the credit but please don't put my name/address there in the body of the document; in the release notes or in the contributors list is fine.

Done; I'll do this:

```latex
In addition to the legacy ports, since release 1.0.3, logic ports following the recommended geometry 
of distinctive-shape symbols in IEEE Std 91a-1991 Annex A (Recommended symbol proportions) 
are also available\footnote{Thanks to Jason for proposing it and digging out the info, see this 
\href{https://github.com/circuitikz/circuitikz/issues/383}{GitHub issue}.}.
```

is it ok for you?

Rmano, 2020-04-17:
[ieeeports-manual.pdf](https://github.com/circuitikz/circuitikz/files/4492627/ieeeports-manual.pdf)

I think it's more or less ready. Will start to prepare the PR...

Rmano, 2020-04-18:
Force-pushed to correct a typo in a commit (I hate them) and I will merge it --- It's easier to test that way. I hope there is no big bug left...

Rmano, 2020-04-18:
Ok --- Please test and reopen if needed

closed 2020-04-18 by Rmano

shirriff, 2021-07-20:
@jason-s: I was looking for "official" information on gate symbols and your research here was very helpful.

minystreem56, 2024-02-23:
I am so sorry, I have an unrelated comment I am trying to make logic gates for my own project, but I am confused by the bubble size can you please clarify that? Other wise my discord is minystreem56, sorry for commenting on this old topic but I don't know any other way to contact the OP.

Rmano, 2024-02-23:
Welcome @minystreem56!  Your best option is to prepare a minimum compilable, full-contained example (see https://tex.meta.stackexchange.com/questions/228/ive-just-been-asked-to-write-a-minimal-working-example-mwe-what-is-that) and post a question on tex.stackexchange.com with tag `circuitikz`. You can browse the Q&A here: https://tex.stackexchange.com/questions/tagged/circuitikz?tab=Newest 

I am sorry, but "I am confused about the bubble size" is not an answerable question, too broad --- I could only point you to the manual about logic ports.

minystreem56, 2024-02-23:
again, I am only asking here becuse this is the only recorce I found for the logic gate sizes, and I do not want to pay $200 for a copy of the standerd

Rmano, 2024-02-23:
The info reported (and linked) is all I have about that. So, I more or less adjusted the size so they look ok. I have no idea if the paid-for standard have more specific info, sorry...

minystreem56, 2024-02-23:
ok! I think the proper size is about 1.5 H

PR #388 (2020-04-18): Added IEEE standard logic gates
Original multi-input code from John Kormylo at tex.stackexchange.com
Help by TheTeXnician <38565529+TheTeXnician@users.noreply.github.com>
Suggested idea and example code by Jason Sachs <jmsachs@gmail.com>
Please see https://github.com/circuitikz/circuitikz/issues/383 for a lot of details
All errors and bug by Romano Giannetti <romano.giannetti@gmail.com>

Fixes #383 

[image]