circuitikz/circuitikz #384: IGBT symbol

RA-EE, 2020-04-07:
I have few suggestions for circuit symbol used for IGBT.

- The present symbol has two identical parallel lines as shown in Fig. 1. However, some standard text books in Power Electronics show the first line (Gate or Base) has to be little shorter compared to the second line. Such a symbol is shown in Fig. 2. I had created it by adding white fill on both sides of the first line. This variation already exists in the nmos shown in Fig. 3. I request the authors to include this new IGBT symbol also in the future versions of Circuitikz.

- Also some users may wish to have higher thickness for the first line and different type of arrows. It will be helpful if it is possible to add customization for the Base-line thickness and arrow tip type.

[IGBT.pdf](https://github.com/circuitikz/circuitikz/files/4443184/IGBT.pdf)

Rmano, 2020-04-07:
This is the image: 
[image]

Rmano, 2020-04-07:
The shorter floating base with adjustable thickness should be easy.

The arrow thing, on the other hand, is quite complex --- they are not real arrow tips for a mainly historic reasons, and if I change it, the idea would be to change it for all the transistors, so that would be quite a big task. Anyway, if anyone is interested... ;-)

Rmano, 2020-04-07:
I have it, using the options:

```latex
\ctikzset{tripoles/igbt/outer base height=0.3}
\ctikzset{tripoles/igbt/outer base thickness=1.5}
```

You have (comparing the normal ones, above, and the modified ones, below):

[image]

Rmano, 2020-04-07:
I will merge this one as soon as I can --- so you can use it with the `git` version. 

The "arrow" part is much more complex; if you want you can add an issue for tracking, but, well, don't hold your breath...

PR #386 (2020-04-08): Added configurability to IGBT outer base
Suggested by Ravindranath Adda <ravindranath@iitg.ac.in>
Fixes #384 

[image]
closed 2020-04-08 by Rmano