circuitikz/circuitikz #385: Improve european logical NOT gates to match IEC 60613-12

j-hap, 2020-04-07:
Hi,

when choosing european logical gates I'd like to have the symbol according to IEC, which has a little circle on the output for NOT, NAND, NOR, XNOR. The current symbols with the little triangle on the port is for indicating the polarity, according to the Standard, the circle is used for negation.
You can see an example of an IEC-NOT here:
https://de.wikipedia.org/wiki/XNOR-Gatter#/media/Datei:IEC_XNOR_label.svg

Thanks for the package!

Regards,

Julian

Rmano, 2020-04-07:
I know *nothing* ;-) about european gates... somebody else can chime in? @mredaelli @sistlind @der-stefan @jesseopdenbrouw ?

jesseopdenbrouw, 2020-04-08:
As far as I know, the circle is used when presenting a logic schematic without any ic type numbering (i.e. just a logic schematic) and the triangle is used when indicating a specific ic type (i.e. when placing a gate on a PCB).

j-hap, 2020-04-08:
I can use this quick workaround. My LaTex skills are a bit rusty, but the package manual helps:
```
  \draw node[european buffer port] (A) {};
  \node at (A.bout) [ocirc, right]{}
```
Which is good enough for me for now.
Suggestion: since the european blocks are mostly the same, you could e.g. add [iec not/nor/nand port] to the possible options, to get the circle and keep the "european" behaviour.

Rmano, 2020-04-08:
Yes, I was thinking about using an option similar to what you did. Although it's not so easy, because it has to twiddle the anchors in case someone chose to a zero-length external pins.
Will look into it!

Rmano, 2020-04-17:
I will tackle this as soon as I merge the IEEE standard ports PR.

Rmano, 2020-04-18:
This is my work-in-progress:
[image]

PR #389 (2020-04-18): Added "not circle" configurability to european port
Suggested by j-hap on GitHub
See https://github.com/circuitikz/circuitikz/issues/385
Fixes #385 

[image]
closed 2020-04-18 by Rmano